ArduPilot/ardupilot #8661: SITL: AC modes Acro and Sport does level

AndKe, 2018-06-16:
In SITL (using USB joystick input) , Acro and Sport levels does autolevel as if they were "Stabilize".
Sport also ignores ANGLE_MAX , and goes way over it's setting.

geofrancis, 2018-06-16:
have you disabled acro trainer?

AndKe, 2018-06-16:
@geofrancis Of course, you are right, ACRO , were just limited by ACRO_TRAINER.

So, the only actual issue is why Sport is not limited by ANGLE_MAX

IamPete1, 2021-08-30:
You blow right past angle max and fly upside down if your `ARCO_RP_P` is big enough.

rmackay9, 2021-08-31:
I don't think Sport mode should be limited by ANGLE_MAX actually.  It is essentially Acro mode with altitude control.  I would actually like to remove this mode because I think it is not very useful.. at least not many people seem to use it.

closed 2021-08-31 by rmackay9