ArduPilot/ardupilot #8663: Ubuntu 18.04 arm-none-eabi toolchain isn't working

khancyr, 2018-06-16:
#### Issue details
On Ubuntu 18.04 toolchain for arm board isn't working. 
It compile fine but board get stuck... 
Please download manually the toolchain from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
Version Version 7-2017-q4-major is working

#### Version
_What version was the issue encountered with_

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

rewolff, 2018-07-17:
FYI. My own project (not ardupilot) also doesn't work on ubuntu 18.04 (with the provided toolchain).

khancyr, 2018-07-17:
yep I think ubuntu arm-none-eabi default toolchain is broken somewhere (again..)

Davidsastresas, 2018-07-17:
I am on ubuntu 18.04 and the toolchain installed with the .sh script on the code ( the 7 version, the one you are pointing to ) works fine. That is the way it is provided in the dev wiki.

rewolff, 2018-07-17:
I used the most recent, 2018Q2 version. Works too. 
I like the situation where more-and-more tools I use are being integrated into Ubuntu so that I don't have to go and install/compile all of the tools I use. So on Ubuntu 14.04 I had to go and fetch the arm-compiler, on 16.04 it was integrated into ubuntu and much less of a hassle, and now we're back to the hassle again. 
Anyway, the default download there is just a tar.bz2 and no instructions on what to do with it. At least I spent 5 minutes looking around for those instructions. Those will very likely be somewhere, but not where I was able to find them. After adding the PPA, it didn't want to upgrade to the new version (because in the end the name was different) an when it did upgrade, there was a conflict with my AVR compiler. Sigh.

WickedShell, 2018-08-26:
@khancyr What resolution are you looking for on this one? We don't actually ever tell anyone to grab the cross compiler from the APT repo, and instead point at a specific version of the tool chain instead. (Cross compiling on anything other then 4.9 or 6.3 is considered a risk and exercise for the user, it should work but it's not regularly tested, and not strictly on the list of things we are trying to maintain).

rewolff, 2018-08-26:
For you as a maintainer for a "small-user-group" project, there is not much you can do. 
But in general it would be nice that Linux evolves from "something for those nerds" to "something everybody can use". And if you have to keep telling people: "this only works with the compiler version X.Y.Z", things are not for everybody. You apparently don't have projects that require the use of compiler version Y.Z.W. And when you do, YOU know enough about YOUR project that you're in a good position to simply make your project work on Y.Z.W and then your project also works on Y.Z.W. 

But for average people that is not an option. So... when things are becoming easier and easier because more and more infrastructure is being added to the repositories, projects should aim to work with the compilers available by default. 

So, when you don't have enough time to install Ubuntu 18.04 (in a virtual box?) and figure out what's wrong, it would be best for the Ubuntu users and ardupilot users if an Item on the todo-list would state: "Figure out why ubuntu arm-none-eabi-gcc doesn't work". 

Most likely it's not YOUR project (most of my projects are affected too), but knowing what's wrong helps everybody. And maybe ardupilot is a good project where the error is obvious enough that it can be isolated....

Progress so far: 
* It is not the compiler (!!!!)
* It seems to be the linking stage. I have compiled a project with an older compiler and the standard one, and exchanging the object files will keep the problem with the linker. 

I currently suspect that it's a library that is linked that causes the problem, not  the linker itself.

WickedShell, 2018-08-26:
Part of my point here is that the default experience we provide is with an install script that provides them with the correct compiler, installing via APT is going out of your way.

Issues with other versions of the cross compiler aren't unheard of, and not unreasonable. I'd embrace things that fix it (if they don't bloat the build size massively, which is sometimes a problem when changing compilers).

Problems at link stage are different then the initial issue reported, a more detailed set of reports on what the error is would help.

khancyr, 2018-08-27:
@WickedShell I made this just to warn user and prevent multiple error report on this issue.
I agree that we should recommand to user to use only the one provide by the install script.
But I find gcc7 more explicite on error so it is better for coding IMHO.
I won't work on a fix for this issue as it is problably an error from ubuntu side.

OXINARF, 2018-09-05:
I'm closing this:

- it's impossible for us to make sure every compiler version works; we provide the exact compiler we use for doing our builds, if a developer doesn't want to use it, he'll need to fix the issues (and contributions for this are welcome!)
- using the Ubuntu provided package is a bad idea, specially when the manufacturer provides its own version of the toolchain and they don't follow it

closed 2018-09-05 by OXINARF