ArduPilot/ardupilot #8668: FMUV2 chirps normal on usb boot... but without SD card

Naterater, 2018-06-17:
#### Issue details
My pixhawk 2.4.8 boots with the normal chirps even though it has no SD card. I remember that without the SD card, the boot chirps should be different

#### Version
Plane chibios 3.9 beta 1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
benchtop test

#### Hardware type
pixhawk 2.4.8

#### Logs

proficnc, 2018-06-17:
Try with USB plugged in

Naterater, 2018-06-29:
The chirps are not the same as NuttX. The chibios tone is similar, but it doesn't change when there is no SD card inserted.

WickedShell, 2019-04-09:
You duplicated your own issue ( #9215 ). Closing this one as the other has more discussion.

closed 2019-04-09 by WickedShell