ArduPilot/ardupilot #8669: Parameters over telemetry very slow when armed

Naterater, 2018-06-17:
#### Issue details
If I ever connect to my plane when it's already armed, the parameter transfer is 10x slower than when it's disarmed.

#### Version
Plane 3.9 beta 1, but I remember this being an issue in the past
Mission planner - many releases

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane, may be others
[  ] Rover
[  ] Submarine

#### Airframe type
believer

#### Hardware type
Cube, RFD900X or 3DR radios

#### Logs

WickedShell, 2019-04-09:
Is this still valid? MAVLink message intervals have experienced a significant rewrite (although not in stable releases I believe) since this was opened. The only thing that comes to mind is if you overly request to many tuning/navigation messages that we only send when in auto/flying you could DOS your own parameter streams a bit.

Anyways I think this is fixed, if I don't hear back I'll go ahead and just close this.

IamPete1, 2021-07-26:
@WickedShell didn't hear back, so I'm going to close.

closed 2021-07-26 by IamPete1