ArduPilot/ardupilot #8679: Plane: building for APM2.6

jackcarrozzo, 2018-06-18:
Hi,

In master, APM1 and 2 aren't in the build list waf gives. I see branches of the project for Plane 3.8.0 and Plane 3.9.0, but the docs suggest the latest Plane version that will run on APM is 3.3.0. 

Is there a branch somewhere that still builds for APM2.6? Or a doc that describes pairing down and building the later Plane versions for APM2?

magicrub, 2018-06-18:
8bit APM arduino hardware was depreciated many years ago. It does not build with waf. Last commit on the last release was 3 years ago on branch ArduPlane-release-AVR. Code to compile it has since been removed since it held back future work. Please upgrade your hardware.

closed 2018-06-18 by magicrub

hiro2233, 2018-06-29:
:thinking:

jackcarrozzo, 2018-07-02:
For what it's worth, I was able to build the older code and make my mods to it for APM by checking out the older tag and figuring out the Makefile.

@magicrub , is there a list of STM32 flight controllers with supported hardware somewhere? I went through the docs and ended up ordering a Beaglebone Blue for $80 (not STM32 obv); if there are cheaper alternatives, I was unable to find them.

It's a shame but understandable that the $40 8-bit controllers are no longer supported, but if there are STM32 options available in that price range it would be exceptional.