ArduPilot/ardupilot #8680: No mavlink UDP connection TX1/ApSync chibiOS

rbachtell, 2018-06-18:
#### Issue details
After flashing chibiOS communication stream was lost over UDP, revert back to nuttx and restored. Cube AP and TX1 with ApSync/OpenKai.  Installed Edison/ApSync and communication as expected   with chibi 
#### Version
Arducopter 3.6rc2

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Hex

#### Hardware type
Cube black

#### Logs
Did not fly with chibi, reverted to nuttx as a safety precaution.

peterbarker, 2018-06-19:
On Mon, 18 Jun 2018, rbachtell wrote:

> After flashing chibiOS communication stream was lost over UDP, revert back
> to nuttx and restored. Cube AP and TX1 with ApSync/OpenKai. Installed
> Edison/ApSync and communication as expected with chibi

Please try 1500000 on serial

rbachtell, 2018-06-19:
Changing serial2 baud rate had no effect

rbachtell, 2018-06-20:
Flashed apsync image to my TX2 and all is well, communication established with chibiOS.  Still can’t get the TX1 to cooperate but it most likely is on my end. No need for anyone to spend any time on this so I’m closing the issue.
Thanks to everyone for the help.      RB

closed 2018-06-20 by rbachtell

rbachtell, 2018-06-21:
Update: I did not go far enough on the baud rate change.  I changed the baud rate in the apsync configuration files as well and was able to get mavlink communication with the TX1/chibiOS.
Thanks again for the help
Cheers, RB