ArduPilot/ardupilot #8681: Rover: compass learning may not be working

rmackay9, 2018-06-18:
This issue came up in [this rover discussion](https://discuss.ardupilot.org/t/compass-learn-in-rover-how-to-use-it).

As mentioned in [this other issue](https://github.com/ArduPilot/ardupilot/issues/8673), rover is saving the compass offsets but so far in testing, it seems that they are never changing.

So this issue involves trying to determine if the compass offsets are being updated ever and if they are not, then discover and fix the cause.

As a side note, in Copter, we manually copy over the offsets from the EKF ([see init_disarm_motors function in ArduCopter/motors.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/motors.cpp#L252)).  It's not clear (to me) why this is required in Copter but not in Plane.  It's also not clear if we should have this in Rover.

apache405, 2019-04-23:
@rmackay9 does this issue still happen on rover?  I have a 250lbs machine that is acting up while we are trying to get the learning cycle done.

rmackay9, 2019-04-23:
@apache405, I'm afraid I haven't investigated this issue yet.  Hopefully when we start beta testing Rover-3.6 we can figure out what's going on.

apache405, 2019-04-25:
@rmackay9 understood.  As we discussed on Skype, I was able to get the compass to calibrate with reasonable offsets of putting the bracket where the cube/here+ lived out of the vehicle and calibrating it outside of the vehicle.  I am looking forward to the compass learning items getting worked out fully in rover 3.6.

rmackay9, 2019-11-05:
Rover-4.0 has a new method of doing "in-flight" compass learning and it has been tested successfully by at least one user so I will close this issue if that's OK.

closed 2019-11-05 by rmackay9