ArduPilot/ardupilot #8686: SET_POSITION_TARGET_LOCAL_NED fails to report currently assigned local target

majorpr13, 2018-06-19:
### Bug report
It appears that when an operator sends the SET_POSITION_TARGET_LOCAL_NED message there exists no acknowledgment of the command. This includes the report via POSITION_TARGET_LOCAL_NED. The ardupilot code overloads the virtual function imposed in the gcs_common for the SET_POSITION_TARGET_GLOBAL_INT variety and has the report implemented for the POSITION_TARGET_GLOBAL_INT, however, appears to not implement the local one. 
Previously discussed on the forum here: 
https://discuss.ardupilot.org/t/set-target-position-return-value/30066/3

**Version**
Arducopter 3.6

This could be an issue in all, however, have only confirmed in SW via the copter and plane branches. 
**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[X] Plane
[  ] Rover
[  ] Submarine

rajat2004, 2019-02-15:
I would like to take up this issue. I'm getting familiar with ArduPilot codebase and SITL testing and solving issues like this will be a big motivation for me.

rmackay9, 2019-02-15:
@rajat2004, sure go for it!

khancyr, 2019-02-15:
I am not really fan of getting a return from setposition, as when doing targeting in speed or position at hight rate, we will spam back the sender

peterbarker, 2019-02-15:
On Thu, 14 Feb 2019, Pierre Kancir wrote:

> I am not really fan of getting a return from setposition, as when doing targeting in speed or position at hight rate, we will spam back the sender

On POSITION_TARGET_LOCAL_NED:

       <description>Reports the current commanded vehicle position, 
velocity, and acceleration as specified by the autopilot. This should 
match the commands sent in SET_POSITION_TARGET_LOCAL_NED if the vehicle is 
being controlled this way.</description>

That seems reasonable to me....  esp. as we can control the message 
rate...

Presumably we would only send this in GUIDED / GUIDED_NOGPS modes?

rmackay9, 2019-02-16:
@peterbarker, maybe we should only send it if the caller has requested it using message-interval.. and if they've done that we send it in all autonomous modes which use a position or velocity target?

peterbarker, 2019-02-16:
On Fri, 15 Feb 2019, Randy Mackay wrote:

> @peterbarker, maybe we should only send it if the caller has requested it using message-interval.. and if they've done that we send it in all autonomous
> modes which use a position or velocity target?

I like the sound of that.

rajat2004, 2019-02-26:
Sorry for responding so late, had some urgent work and then was travelling.

So should it send back the newly assigned target using POSITION_TARGET_LOCAL_NED or return the result of the command using MAV_RESULT?
In the latter case, there could be different messages returned like MAV_RESULT_ACCEPTED or MAV_RESULT_DENIED depending on the situation.

rajat2004, 2019-03-02:
Is there any way to directly get the current POSITION_TARGET_LOCAL_NED data similar to the ```target``` object in the following function 
[https://github.com/ArduPilot/ardupilot/blob/038dd68e3a69388e9dc1ac6dde7ff3aaaf549934/ArduCopter/GCS_Mavlink.cpp#L97](https://github.com/ArduPilot/ardupilot/blob/038dd68e3a69388e9dc1ac6dde7ff3aaaf549934/ArduCopter/GCS_Mavlink.cpp#L97)

I also tried looking at the implementation of [POSITION_TRAGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#POSITION_TARGET_LOCAL_NED) (```MSG ID 85```) but I wasn't able to find it.

peterbarker, 2019-03-03:
@rajat2004 This is probably a can of worms you're opening here.  I *believe* `POSITION_TARGET_LOCAL_NED`  basically reflects back the values from the  equivalent SET.  That's a problem if someone else has played with pretty much anything.

Knowing which of these controllers is active would probably be the first step:
https://github.com/ardupilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L1071

rajat2004, 2019-03-05:
While searching for the implementation of the POSITION_TARGET_LOCAL_NED, I found [the header](https://github.com/mavlink/c_library_v1/blob/master/common/mavlink_msg_position_target_local_ned.h) file for the message in Mavlink, in which there's a function [```mavlink_msg_position_target_local_ned_send```](https://github.com/mavlink/c_library_v1/blob/5ce595557767fdf2e13ad8190e499fb53e2e92f1/common/mavlink_msg_position_target_local_ned.h#L260) similar to the one used to send POSITION_TARGET_GLOBAL_INT [here](https://github.com/ArduPilot/ardupilot/blob/038dd68e3a69388e9dc1ac6dde7ff3aaaf549934/ArduCopter/GCS_Mavlink.cpp#L97)

So what I thought is that we can use the ```mavlink_msg_position_target_local_ned_send``` function at the end of [```case MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED```](https://github.com/ArduPilot/ardupilot/blob/038dd68e3a69388e9dc1ac6dde7ff3aaaf549934/ArduCopter/GCS_Mavlink.cpp#L992) in the GCS_Mavlink.cpp file like this:
```
mavlink_msg_position_target_local_ned_send(
            chan,
            AP_HAL::millis(),
            packet.coordinate_frame,
            packet.type_mask,
            packet.x,
            packet.y,
            packet.z,
            packet.vx,
            packet.vy,
            packet.vz,
            packet.afx,
            packet.afy,
            packet.afz,
            packet.yaw,
            packet.yaw_rate);
```

I added this in the file and built the firmware, it built successfully and passed the tests also. I then tested this in SITL, the copter is receiving the message but I'm still not receiving any POSITION_TARGET_LOCAL_NED messages in the listener.
Is there any other way to check the messages other then dronekit?

Also, I'm getting the following message in the console while testing

[image]

@peterbarker I wasn't able to find anywhere if the type_mask or the data of the message is being stored somewhere. If it was being stored, this could have been done in a different function. Please correct me if I'm wrong.
That's why I added the function at the end of SET_POSITION_TARGET_LOCAL_NED so that we can send back the message if the target has been set, with same data.

khancyr, 2019-03-05:
hello,

we don't store the data from those message.
to see your message, on mavproxy  console type : watch POSITION_TARGET_LOCAL_NED

rajat2004, 2019-03-05:
Thanks a lot for the info, I just tested it with mavproxy and I'm receiving the POSITION_TARGET_LOCAL_NED messages. Not sure why Dronekit isn't listening to this message.

For the GCS, I'll have to experiment a bit since I'm currently testing everything inside a Docker container due to my system installation being a bit unstable right now.

Any idea as to why the red messages are popping up in the console?

khancyr, 2019-03-05:
surely a logging function that isn't write correctly

rajat2004, 2019-03-07:
@peterbarker I searched for the cause of this problem and found that it was being caused by [ArduPilot/ardupilot#10669](https://github.com/ArduPilot/ardupilot/pull/10669) which was merged just a few days back.

Can u guide me as to what changes I should make to fix this problem?

peterbarker, 2019-03-08:
@rajat2004 I'm going to  bring that problem up on the devcall.  The sanity checks says that a floating point number shouldn't have a multiplier - so we should store degres, not centidegrees in a float field, for example.  The warning is benign - you can ignore it.  If you feel like creating a PR to remove the multipliers (and store the data in the base-SI unit), that would give a talking point for Tuesday :-)

rajat2004, 2019-03-08:
I have opened a PR for Copter only as of now. I'll add the same for all the other platforms after feedback and changes.

rajat2004, 2019-04-09:
Function for sending the current local target has been added. The operator can now request the target using SET_MESSAGE_INTERVAL if required. There hasn't been any addition such as to automatically send back the target info at a constant rate and this can be decided by the operator if the data is required or not. 
Is there any further action which needs to be done for this issue or should this be closed?

peterbarker, 2019-04-09:
All done on this one.

Note that I addressed a small issue today where we were sending a zero-throttle input when we switched to loiter mode.  This actually seems to have uncovered a bug in RTL... so by patching the bug in the test I've hidden that bug again...

As for other vehicles, Rover could use it... Plane doesn't use set-position-target-local-ned for anything sensible...

closed 2019-04-09 by peterbarker

rajat2004, 2019-04-09:
Oh, I had no idea that adding new tests would reveal bugs in other unrelated things. Shows how much more I have to learn.
@peterbarker Thanks a lot for fixing the tests. And for guiding me on how to contribute and merging my contributions.

I'll look into Rover, see if this can be added

oceanalan, 2019-06-05:
@rajat2004 rajat2004 I am using the SET_POSITION_TARGET_LOCAL_INT with the Rover, and do need feedback for this. Has anyone integrated this fix into the Rover yet ?

rajat2004, 2019-06-06:
@oceanalan The support for POSITION_TARGET_GLOBAL_INT messages for Rover was recently added by @rmackay9 in https://github.com/ArduPilot/ardupilot/pull/11420
This has been merged and should work with SET_MESSAGE_INTERVAL also