ArduPilot/ardupilot #8687: ArduCopter EKF position differs from RTK GPS Position up to 60cm.

amilcarlucas, 2018-06-19:
### Bug report

ArduCopter EKF position differs from RTK GPS Position up to 60cm.

**Issue details**

The EKF position innovations are in the range of -20 to 20 cm.
I would expect this value to be much lower.
We use a Septentrio AsteRX-m2 at 10Hz. I've set the EK2_GPS_DELAY to 55 ms.
The flight speed is mostly constant at ~5 m/s the flight path is a circle.

When Vx is aprox 0 the X position innovations are also almost zero.
The larger the Vx the larger the X innovations.

The same on the Y axis. Small Vy -> small Ypos innovations, the bigger the Vy the bigger the innovations.
So it looks like at small speeds the EKF does a good job.
At 5m/s it just fails miserably.

When comparing GPS positions with EKF positions we used this formula:
(GPS2.Lat-Pos.Lat)*111275.93
(GPS2.Lgn-POS.Lgn)*67980.245
And see that they differ up to 60cm from each other.

How can the difference between EKF and GPS grow from zero to up-to 60cm within 100ms ??
The copter is flying at ~5m/s that is 50cm/100ms. 

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
3DR X8+

**Hardware type**
Pixhawk 2.1

**Logs**
https://transfernow.net/9109g1l28wks

WickedShell, 2018-06-19:
Couple of things:

Where did the 55 ms number come from? (Last time I did a gps lag estimate on a AsteRx-M it came out to 80 ms, not 55).

Your AHRS and AsteRx-m2 position should disagree, you've loaded a (0.11, 0, -0.125) meter offset.

The AsteRx-m2 is not your primary GPS, the u-blox is, so we don't expect to see this match perfectly.

Closing this out as a support issue to be better tracked on discuss.

closed 2018-06-19 by WickedShell

amilcarlucas, 2018-06-20:
Yes, the forum is probably a better match: I have placed it https://discuss.ardupilot.org/t/ekf-position-differs-from-rtk-gps-position-up-to-60cm/30158 and I answered your question there as well.