ArduPilot/ardupilot #8721: Add support for Angle of Attack (AoA) sensors

MacBr, 2018-06-23:
### Feature request
**Summary:**
I would like to add support for magnetic angle position sensors using the I2C input. These would be used in vane type instruments to determine angle of attack (AoA) and side slip angle (SSA). The issue of angle of attack sensors has been raised in the past but the issue was closed without resolution and additional work should be a new issue (@magicrub). The sensor we have is the AS5600 from AMS which has I2C output, and would like to use this this digital output (reasons below). Thus, the drivers and necessary front-end code alterations would need to be made.

I am not great at C++ and I am just starting out with using ArduPilot/MissionPlanner and the task seems to be above me so any help would be greatly appreciated, let alone improve the impressive capabilities of this open source project, kudos to all! I will also be posting the results of testing and calibration as well as details/plans of our instrument setup for others to use on their projects.

Previous unresolved AoA hardware issue:
#2928
Software (AoA estimation):
#5933

**Details:**
First some background. This AoA and SSA sensor will be used on (large) research aircraft investigating the feasibility, handling and performance of new concepts. I work as part of a team at a good institute and it is the aim to achieve as accurate results as possible, naturally. Accurate AoA and SSA measurements are critical, especially if the results are to be scaled up to the full size aircraft (transonic transports). The setup will be tested and calibrated in wind tunnels, alone and on aircraft, as well as flight tested.

There could be a discussion as to the type of AoA sensor but this has been done internally and decided to go ahead with a vane type instrument. The magnetic angle sensor described is well suited to this task, @Georacer has already made a good [report](https://discuss.ardupilot.org/t/low-cost-air-data-instruments-calibration-and-discussion/16024) detailing the use of a similar sensor but using the analogue output. Another research team here at our institute has also used this sensor on a PixHawk in analogue and gotten good results. They however recommend using the digital output and would also benefit from this upgrade. This is even more the case for the current aircraft as cable lengths are longer and there are likely more EMI sources.

**What is needed:**
I have all of the required hardware to test with. I need the software implementation for the sensor into ArduPlane and MissionPlanner. My C++ skills would not allow me to come up with **good** code in any decent amount of time. In my opinion it would be best if one of the devs could look at this. I am willing to help wherever I can and all of the required information about the setup is known. For info, in MissionPlanner it would be nice if the AoA and SSA are shown (like the currently estimated values) but in principle they will be logged and used for post-processing.

**Setup:**
PixHawk 2.4.6
ArduPlane 3.8.5
MissionPlanner
Other I2C devices:
I2C splitter
MS4525 and SDP3X airspeed sensors (dual and alone for calibration/accuracy determination)
Holybro NEO-M8N GPS with HMC5883L compass (I2C)
2x AMS AS5600 magnetic angle sensor for AoA and SSA

**Issues:**
Two angle position sensors for AoA and SSA using the same address, the AS5600 chip is fixed at 0x36. Yes this is an issue but for now we will only use one at a time, or eventually we might use an address translator (like LTC4316). Plan B is use the SSA chip in analogue mode and the analogue input on the PixHawk. If someone has a good idea for us regarding this please let me know. There should not be other _address_ issues with the setup.

**Resources:**
As far as I know all of the I2C properties needed to make the driver can be found in the AS5600 datasheet:
[ams.com/documents/20143/36005/AS5600_DS000365_4-00.pdf/555ac681-9a88-be5e-a615-c1aecfb27edb](https://ams.com/documents/20143/36005/AS5600_DS000365_4-00.pdf/555ac681-9a88-be5e-a615-c1aecfb27edb)

palm369, 2018-10-25:
This is a fantastic idea and I support this! But I am no coder myself, I can only support with flight tests and custom built sensors.

My plan is to put two vanes on a carbon stick, each vane with a magnet inside and an hall sensor next so the magnet on the stationary part. Then I read the hall sensors with an good ADC with is connected to an arduino. The arduino then can calculate the SSA and AOA and output this as Serial, I2C, PWM or analog.
Hardware cost <20 $

Another option is your sensor. Maybe its even better.

moreba1, 2020-12-27:
https://www.swiss-airdata.com/products/smv1 
angle of attack sensor for UAV

peterbarker, 2021-01-21:
https://github.com/ArduPilot/ardupilot/pull/16396/files will close this eventually.

tekcadi, 2024-01-24:
Hello guys, is there any news on this?
I'm thinking of building an AP_AngleofAttack library to interface with the SwissAir Smv sensor, currently developing and interface board to make the sensor communicate over UAVCAN.