ArduPilot/ardupilot #8724: Plane: head towards next waypoint instead of waypoint lines

expntly, 2018-06-24:
### Bug report

**Issue details**

Plane flies towards waypoint "lines" as opposed to towards next waypoint.  This is what happens in AUTO:

[image]

It seems like the right thing to do would be to have the plane head directly towards the next waypoint.  This would make for a more optimal flight path, and avoid having the plane roll left and right all the time unnecessarily.

**Version**
All versions, latest tested is 3.8.5beta1.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QP X8

**Hardware type**
PH2.1, reproduced on SITL.

WickedShell, 2018-06-24:
It depends upon your vehicle/mission. The current behavior is much better for mapping missions, and is what you see manned aviation doing (and generally is more predictable where the vehicle will travel). We could potentially add a flag to change that behaviour, but it wouldn't be a reasonable default behavior. (I'm generally struggling to find a situation that is best served by direct routing/not fighting the effect of wind)

magicrub, 2018-06-24:
This is by design. It's called cross-track (or xtrack) and it's how to fly a planned path. The problem you're having is your NAVL1_PERIOD is too big for your aircraft (or some other turn limit) so your turns are very long.

If this is turned off, your flight would be very sloppy and unpredictable. You would also drift around in the wind.

However, for loiters there is an option to exit on the tangent and to xtrack from the exit point instead of the center point. Here's an example:
[image]

@WickedShell suggested we could add this to normal waypoints too but I'm not sure at what point we would "grab" the point to xtrack from. Maybe just turn off xtrack completely and let it drift in the wind? Hmm.. sounds very sloppy.

fix your turn problem and report back

closed 2018-06-24 by magicrub

WickedShell, 2018-06-24:
@magicrub the plane can fly with no xtrack at all (that's how paths to initial waypoint is computed/changes into auto from any other mode), but effectively what happens is it just aims at the waypoint, accepts being blown sideways by wind, and then just adjusts the aim point to be at the waypoint again.

expntly, 2018-06-25:
That

[image]

is exactly what I'm looking for.  Except for pure waypoints.

I'm not sure I'm following you both on why this is available for loiter stuff but not regular waypoints.  Why does it make sense in the case of loiter type wpts?

I'll try the NAVL1_PERIOD in the meantime as suggested.

magicrub, 2018-06-25:
Your aircraft turns are being limited. Also check LIM_ROLL_CD and others..

expntly, 2018-07-13:
Tried NAVL1_PERIOD today, from 20 to 18, nothing changed.  The document says it should be lowered for smaller aircraft, but our aircraft is 18ft wingspan, so I'm hesitating to lower it even more.

LIM_ROLL_CD is at 25.  I don't see how adjusting this will disable xtrack, which is what I'd like ultimately.

Also tried different values for `NAVL1_XTRACK_I`, 0.2 to 0.1 to 0.05 to 0.01, no difference.

@magicrub Any suggestions?  Thanks