ArduPilot/ardupilot #8735: RC Cal Safety/Usability to prevent arming

DonLakeFlyer, 2018-06-25:
Would be awesome if ArduPilot supported MAV_CMD_PREFLIGHT_CALIBRATION.param4=1/0 to start stop radio calibration. Mavlink spec is not super specific by in QGC this causes the motors to not arm when you are doing RC cal. This way you don't have to take props off during RC cal.

closed 2021-11-01 by tridge

IamPete1, 2021-11-01:
@DonLakeFlyer AP should now support this, is there any change required to QGC to enable sending the message? I could not seem to get it to send.

DonLakeFlyer, 2021-11-02:
I'll take a look. It's a bit tricky since it's firmware version specific. But good safety thing so I'l try to get it supported in this next stable which I'm working on right now.

DonLakeFlyer, 2021-11-02:
I looked, yes this does require a QGC change. It's currently not sent for ArduPilot only PX4.

peterbarker, 2021-11-03:
On Mon, 1 Nov 2021, Don Gagne wrote:

> I looked, yes this does require a QGC change. It's currently not sent for ArduPilot only PX4.

Hopefully this is just removing special-case code :-)

DonLakeFlyer, 2021-11-04:
> Hopefully this is just removing special-case code :-)

Not quite. What's going to be the version number for firmware for which this command is supported? I need to special case to that version since it's the only way to know if this is supported or not.

IamPete1, 2021-11-04:
Should be OK to go with 4.2 and newer. Unfortunately we were using the param incorrectly in the past, so sending it on older versions will do the wrong thing.

DonLakeFlyer, 2021-11-04:
> Unfortunately we were using the param incorrectly in the past, so sending it on older versions will do the wrong thing.

Oh, crap. That means my pull will be wrong for versions < 4.2

DonLakeFlyer, 2021-11-08:
@IamPete1 Isn't 4.2 the current stable? If so it seems like this change just went in a few weeks ago which would seem to mean it's not in stable until 4.3?

IamPete1, 2021-11-09:
@DonLakeFlyer 4.1.x is current stable. 4.2.dev is master.

DonLakeFlyer, 2021-11-09:
Crap sorry. I was off by one! Anyway I'll fix up QGC to only turn this one when running 4.2+. Right now the code does it for all versions which is bad since older firmwares do something odd with it.

IamPete1, 2021-11-09:
@DonLakeFlyer Thank you

PR #18514 (2021-11-01): Fail to arm if in RC_Calibration
Fixes #8735

Although as far as I can tell QGC no longer sets that bit. I think it would be a nice enhancement to all the GCSs tho.

I'm not sure on the exact way to use the pram. 

https://mavlink.io/en/messages/common.html#MAV_CMD_PREFLIGHT_CALIBRATION

Says: 
     1: radio RC calibration, 2: RC trim calibration

Currently I have 1 = running 0 = stop. But that means we cant ack the stop. Maybe it should be a continues set and timeout. Or a toggle. 

Since QGC seems not to send it anymore we have no base line to compare to....