ArduPilot/ardupilot #8757: GNSS V2 Barometer "Bad Baro Health"

Rickyg32, 2018-06-28:
#### Issue details
I recently installed a brand new GNSS V2 GPS, which has an onboard barometer. I immediately began getting "Bad Baro Health" messages in MP. I tried all that I could to get it corrected, read all the documentation, searched the forums. No go. I rolled back the firmware on the Pixracer and found that any of the 3.5.x firmwares had this issue and 3.4.6 works fine. Sadly the NTF_Display function is missing from that version so I don't have all my design features working. 3.4.6 is fine for now but thought I would raise the issue. I tried to find it as a bug but no luck sorry if this is a repeat.

#### Version
The newest which I believe was 3.5.3 I think..perhaps 3.5.4..either way none of the 3.5.x versions work

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_ Quad (x format)

#### Hardware type
Pixracer

#### Logs
No logs..will see if I can create one.

Rickyg32, 2018-07-03:
Do you folks need something to get this on the bug list.

rhenetouv, 2018-07-27:
I encountered this problem too with arducopter 3.5.6 and GNSS V2, hardware version 2.2, firmware version 4.0

Rickyg32, 2018-08-29:
Looks like we are out of luck for now

auturgy, 2018-08-29:
A log would be great. Can you also confirm the specific gps?
I suspect that you’re referring to the Zubax UAVCAN GNSS v2, but it isn’t clear.

Rickyg32, 2018-08-29:
Hey auturgy yes it's a GNSS v2 GPS. The GPS appears to work but the Baro gives the bad baro health message. I will see if I have a long, but it was a couple months ago now and I have been using 3.4.6 to make it work so I am not sure I have a log from that firmware. But I will look.
Thank you for responding. There are a few of us with this issue.