ArduPilot/ardupilot #8775: Arducopter 3.6.0RC2 IST8310 compass not found on Omnibus F4V3 board

Ralfde, 2018-06-29:
Omnibus F4V3 Arducopter 3.6.0RC2 hardware with IST8310 connected to I2C2 (at UART3 pins)
(Quadro X)
The compass is not found. What do I need to set, so the compass is found?

I disabled the arming precheck for the COMPASS, because I wanted to test the motor order and arming, but
arming is not possible. Compass not healthy.
Which I2C address is accepted? i already tested 0x0c and 0x0E

What do I need to do to solve this problem?

I have also the RC9-16 trim error message as another user above, but I could solve this by not checking the RC Control at the arming check.

sh83, 2018-06-30:
Only hmc5883l is detected on external i2c bus. Compass and baro autodetection was discussed, but it was not done yet.
You could use hmc5883l compass or i could build custom build with IST8310 compass for you.

Ralfde, 2018-06-30:
The HMC5883L is not available any more. When a firmware is done for a new board, it would be very good, if the actual compasses would be supported.
It would be very good, if you could offer a version for the IST8310. If possible, please make the custom build for I2C address 0x0C or checking both addresses 0x0C and 0x0E. In advance I would like to test a custom build, if possible. Just let me know, which address I need to use. If manual selection would be more easy to do, this would be no problem to deselect the not used ones with mission planner.

sh83, 2018-06-30:
Binaries is here https://github.com/sh83/ardupilot/blob/ist8310/arducopter-test.zip
Build from recent sources with this fix https://github.com/sh83/ardupilot/commit/563442c0a54e259cd28b33170dadc6bd284d0b63
Address is 0x0C

I fully agree that compasses should be autodetected, but I can not do much about it.

Ralfde, 2018-07-01:
Thanks a lot. I will test it.

Ralfde, 2018-07-01:
Thanks. Now it works.

Ralfde, 2018-07-02:
I can't reduce the P and I values enough for a racecopter, because of the minimum limits of the values.
Do you have a working setting for a racecopter? Does it maybe work with a higher PWM frequency?
The copter tries to fly shaking away, because the minimum values are still too high. So I need to cut-off the motors.

auturgy, 2018-07-02:
Support questions should go in the forum.
If you’re using missionplanner, use the advanced view, full param list and set the gains directly. They aren’t min/max checked if you do it that way.

Ralfde, 2018-07-02:
Ok, great. Thanks a lot.

Ralfde, 2018-07-03:
Copter flies good now with RC2. Just the altitude hold is to jumpy. Where can I find the PIDs for the altitude hold? The motors are very powerful and I think I have to reduce the P value by factor 4.

sh83, 2018-07-03:
It should work after tuning RNGFND* params.

Ralfde, 2018-07-03:
Tried to build the RC3 with IST8310 myself, but get this error:
What do I need to do to solve this?

Waf: Leaving directory `/mnt/c/dev/ardupilot/build/omnibusf4pro'
Build failed
 -> task in 'objs/AP_Terrain' failed (exit status 1):
        {task 139732806704336: cxx AP_Terrain.cpp -> AP_Terrain.cpp.0.o}
/home/rh3141sj/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -std=gnu++11 -fdata-sections -ffunction-sections -fno-exceptions -fsigned-char -Wall -Wextra -Wformat -Wshadow -Wpointer-arith -Wcast-align -Wundef -Wno-unused-parameter -Wno-missing-field-initializers -Wno-reorder -Wno-redundant-decls -Wno-unknown-pragmas -Werror=format-security -Werror=array-bounds -Werror=uninitialized -Werror=init-self -Werror=switch -Werror=sign-compare -Wfatal-errors -Wno-trigraphs -Werror=unused-but-set-variable -Wlogical-op -Wframe-larger-than=1300 -fsingle-precision-constant -Wno-attributes -Wno-error=double-promotion -Wno-error=missing-declarations -Wno-error=float-equal -Wno-error=undef -Wno-error=cpp -Wno-cast-align -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wall -Wextra -Wno-sign-compare -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=uninitialized -Werror=init-self -Wframe-larger-than=1024 -Werror=unused-but-set-variable -Wno-missing-field-initializers -Wno-trigraphs -Os -fno-strict-aliasing -fomit-frame-pointer -falign-functions=16 -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -fno-builtin-fprintf -fno-builtin-vprintf -fno-builtin-vfprintf -fno-builtin-puts -mcpu=cortex-m4 -mno-thumb-interwork -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DCHIBIOS_BOARD_NAME="omnibusf4pro" -MMD -include ap_config.h -Ilibraries -Ilibraries/GCS_MAVLink -I. -I../../libraries -I../../libraries/AP_Common/missing -I../../modules/ChibiOS/os/license -I../../modules/ChibiOS/os/common/startup/ARMCMx/compilers/GCC -I../../modules/ChibiOS/os/common/startup/ARMCMx/devices/STM32F4xx -I../../modules/ChibiOS/os/common/ext/CMSIS/include -I../../modules/ChibiOS/os/common/ext/CMSIS/ST/STM32F4xx -I../../modules/ChibiOS/os/rt/include -I../../modules/ChibiOS/os/common/oslib/include -I../../modules/ChibiOS/os/common/ports/ARMCMx -I../../modules/ChibiOS/os/common/ports/ARMCMx/compilers/GCC -I../../modules/ChibiOS/os/hal/osal/rt -I../../modules/ChibiOS/ext/fatfs/src -I../../modules/ChibiOS/os/hal/include -I../../modules/ChibiOS/os/hal/ports/common/ARMCMx -I../../modules/ChibiOS/os/hal/ports/STM32/STM32F4xx -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/ADCv2 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/CANv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/DACv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/DMAv2 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/EXTIv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/GPIOv2 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/I2Cv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/MACv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/OTGv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/QUADSPIv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/RTCv2 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/SPIv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/SDIOv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/TIMv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/USARTv1 -I../../modules/ChibiOS/os/hal/ports/STM32/LLD/xWDGv1 -I../../modules/ChibiOS/os/hal/lib/streams -I../../modules/ChibiOS/os/various/cpp_wrappers -I../../libraries/AP_HAL_ChibiOS/hwdef/common -DSKETCHBOOK="/mnt/c/dev/ardupilot" -DCONFIG_HAL_BOARD=HAL_BOARD_CHIBIOS -DHAVE_OCLOEXEC=0 -DHAVE_STD_NULLPTR_T=0 ../../libraries/AP_Terrain/AP_Terrain.cpp -c -olibraries/AP_Terrain/AP_Terrain.cpp.0.o

Ralfde, 2018-07-04:
Changed from Ubuntu to Cygwin now. Seem to make less trouble.
[image]

But it makes a .bin file. How to update it?
How to make a .hex file with bootloader?

sh83, 2018-07-04:
Just load *.apj file as a custom fw with mission planner.

Ralfde, 2018-07-04:
Thanks a lot, find it out at the moment and works.

Ralfde, 2018-07-04:
Sorry, one more question about the Omnibus F7V2. Does this one detect the IST8310 with 0x0C automatically or if not, which custom build/changes would be needed?

sh83, 2018-07-04:
No it does not. F7 omnibus has 512kb flash,  which is good for betaflight, but not for arducopter.

Ralfde, 2018-07-04:
Ok. Please tell me, what I have to change to make the IST8310 work with F7V2.

sh83, 2018-07-05:
You could apply simliar fix (https://github.com/sh83/ardupilot/commit/563442c0a54e259cd28b33170dadc6bd284d0b63) to the omnibus f7 hwdef.

closed 2018-07-09 by Ralfde

sh83, 2018-07-14:
For you information. Compass autodetection was enabled in recent build

Ralfde, 2018-07-14:
Thanks a lot for this info.