ArduPilot/ardupilot #8787: Rover: add RTK GPS failsafe to handle GPS position jumps when RTK fix is lost

rmackay9, 2018-07-02:
some RTK GPSs (including the Piksi) can suddenly jump the position a few meters when they lose RTK fix.  This request from jcantilo in [this discussion](https://discuss.ardupilot.org/t/rtk-fixed-float-jump-in-rover/30444) is to add a failsafe so the vehicle is switched into Hold mode when RTK fix is lost, and then return to the previous mode when RTK fix is restored

liang-tang, 2018-07-13:
copter really needs it

samuelclavien, 2019-01-10:
Hi, this would be a really important feature for use with RTK GPS because only RTK FIX can be trusted. It would also be good to avoid letting switch to Auto mode if GPS status is not 6 (FIX). Even with pre-arm check set to "3D Fix", an "RTK Float" status is enough to arm the rover for now.

ktrussell, 2019-05-10:
My GPS (U-blox C94-M8P) doesn't seem to jump as described above when losing RKT Fixed and vice versa, but this is a feature I desire. I have plans to try to implement it in a co-processor, but if it could be built-in as described, that would be great. With my mower, I do see missed lines of grass when in float. I would like the rover to simply go into hold if RTK Fixed mode is not present and return to Auto when RTK Fixed returns.

Philippe69630, 2023-08-05:
Hello, I have the same problem with my mower, it will be very nice to HOLD when GPS RTK Float and go back in AUTO when GPS find RTK Fixed.