ArduPilot/ardupilot #8792: AC_Avoid: adjust_velocity_polygon fuction miss boundary[0]

JIMWAR, 2018-07-02:
AC_Avoid::adjust_velocity_polygon miss boundary[0]
`   for (i=0,j……) rather than i=1

my vehicle can't go forward no obstacle ahead and still go left when "i=1"

Copter v3.5.5

rmackay9, 2018-07-02:
Thanks for the report although I think we need some more details to really understand this.

For example more explanation of what is being done and/or a dataflash log file, perhaps a file with the polygon points.

JIMWAR, 2018-07-02:
Sorry, my description is not careful enough.OK, again，more detail is as follow.
when i test avoidance fuction with pixhawk(firmware version:Copter v3.5.5) in loiter mode,the vehicle no obstacle ahead cannot move forward and go all the way to the left.I set  the AVOID_MARGIN parameter as 3 and the flight log file is avoid_loiter(i=1).BIN in the appendix.
I begin to check the code which involve avoidance fuction in loiter mode, i find "adjust_velocity_polygon" function in AC_Avoid.cpp file
[image]
 that the for loop i starts from 1 instead of 0(see adjust_velocity_polygon.png file in the appendix),therefore  i think the boundary[0] is ignored and cause  still forward obstacle avoidance.
After i modify the for loop 'i' to 0,the vehicle forward obstacle avoidance is normal. the flight log file is avoid_loiter(i=0).BIN in the appendix

IamPete1, 2021-07-26:
Looks like this has been fixed.

https://github.com/ArduPilot/ardupilot/blob/f20644b6cf87b53c2394ad1ff07673c5ae657864/libraries/AC_Avoidance/AC_Avoid.cpp#L1300

closed 2021-07-26 by IamPete1