ArduPilot/ardupilot #8800: Rover: support vehicles with lower speed in reverse

rmackay9, 2018-07-03:
Rover-3.4 (and lower) supports driving forwards and backwards but the speed controller assumes that the vehicle's maximum speed backing up is the same as it's maximum forward speed.

We should improve support for vehicles with lower reverse speeds (or vehicles that can't go in reverse at all).

Perhaps we could add a parameter to capture the maximum speed in reverse (SPEED_REV_MAX?).  The problem with adding this parameter though is that we don't have an equivalent for the maximum forward speed.  Instead we extrapolate the vehicle's maximum speed by looking at the CRUISE_SPEED and CRUISE_THROTTLE value.

Perhaps we could add a CRUISE_SPEED_REV and CRUISE_THROTTLE_REV which could be used in place of CRUISE_SPEED and CRUISE_THROTTLE when the desired speed is negative.  I don't really like this proliferation of parameters because it will complicate the setup for the user.

A third idea is to compare the pwm range between SERVO3_MIN to SERVO3_TRIM vs SERVO3_TRIM to SERVO3_MAX.  If we assume the ESC's speed response is linear then if the low range is small, then the reversing speed is low.  For example if the SERVO3_MIN is 1250, SERVO3_TRIM is 1500 and SERVO3_MAX is 2000, then we know that it's reversing speed is 1/2 it's maximum forward speed.

mikerob, 2018-07-03:
A single "reverse gain" would let the user apply a multiplying factor to negative throttle - that may cover most of the cases.  For rovers with IC engines and mechanical brakes, it would allow you to correct for the difference in authority and response between the throttle and brake.

rmackay9, 2024-03-13:
This is handled in 4.4 using the MOT_THST_ASYM parameter so I will close this.

closed 2024-03-13 by rmackay9