ArduPilot/ardupilot #8828: using proximity sensor downward oriented to follow ground level

roque-canales, 2018-07-05:
hello,

I use mavlink prx messages for input to pixhawk distance from ground.

it's working perfectly in prx mode.

But it no work as range finder….

How to set pixhawk for it use this same sensor (orientation 25) for terrain Following?

I attempt to set rngfnd_type to mavlink but this no work…..

I Don't find message type for "mavlink rangfinder" input to pixhawk.

Please let me know how to do

IamPete1, 2021-08-30:
MAVLink rangefinders use the DISTANCE_SENSOR message. That should work for both proximity and rangefinder. Possibly this is what you tested?

https://mavlink.io/en/messages/common.html#DISTANCE_SENSOR

Once it is detected as a rangefinder you will then be able to use it for terrain following.

IamPete1, 2021-08-30:
In fact it looks like we only accept downwards facing MAVLink rangefinders, I'm not sure why...

IamPete1, 2021-09-11:
@roque-canales please re-open if this is still a issue.

rmackay9, 2023-07-04:
This is still a valid enhancement request so re-opening.

rfv-370, 2023-08-01:
I believe the issue is valid for all Mavlink message on all diretcion tested on SITL and physical drone and rover, same issue , as soon as you use Mavlink message, the proximity radar works but not the object avoidance....: https://discuss.ardupilot.org/t/mission-planner-sitl-rangefinder-not-avoiding-obstacles/104008
I suspect it is same issue or shall I open another thread ?

lvale, 2023-08-01:
@rmackay9  couldn't extending the sectors concept used for the Proximity be used to acomplish this?

rfv-370, 2023-08-09:
@rmackay9 and @lvale So I guess Mavlink message for all directions are not working on PRX mode, right:
https://discuss.ardupilot.org/t/mission-planner-sitl-rangefinder-not-avoiding-obstacles/104008/6
So any Mavlink message for Range finder does not its way to the proximity algo??
Would someone point where in the firmware code is the issue so we could a quick work around while this is fixed officially?