ArduPilot/ardupilot #8832: ChibiOS: Add support for stm32F413

magicrub, 2018-07-05:
### Feature request

Please add support for STM32F413. @bugobliterator has added support on these two branches but they conflict with branch:pr-chibios-update

https://github.com/bugobliterator/ardupilot/tree/pr-f413-nucleo
https://github.com/bugobliterator/ChibiOS-1/tree/pr-f413-dma

Existing problem:
current ChibiOS did not have the STM std lib that had f413 in it. It's a new cpu.
ChibiOS just had a release (May 2018) that included the new STM32 std lib, but no chibiOS support. Then a few (2?) weeks later someone added f413 support. I believe Sid's branch is the chibios/upstream/master that includes those 413 hooks.

So, if we are going to switch over to the "latest ChibiOS", please please please include those few extra commits that allow F413

tridge, 2021-08-30:
it would be easy to add F413 if its needed, I don't think we need an issue for it

peterbarker, 2024-07-31:
We haven't needed this in 6 years.  If it comes up we can deal with it then...

closed 2024-07-31 by peterbarker