ArduPilot/ardupilot #8842: Multiple planes getting same waypoints in gazebo SITL

DevonMorris, 2018-07-06:
### Bug report

**Issue details**

I am trying to fly 2 planes in a gazebo SITL environment. To do this, I am commanding waypoints via mavros. Whenever I command a new waypoint table, I clear the previous one using the `mavros/mission/clear` service and then I push my new waypoint table using `mavros/mission/push`. I have namespaced my two vehicles to avoid conflicts in the ros service calls. They are namespaced as `mav0` and `mav1`.

I have two vehicles in my simulation, corresponding to different instances of `sim_vehicle.py`. Explicity, I ran
```
sim_vehicle.py -v ArduPlane -f gazebo-zephyr --map --console --instance 0
sim_vehicle.py -v ArduPlane -f gazebo-zephyr --map --console --instance 1
```
from a ros launch file.

When this is run, I get two different windows displaying MavProxy commands
```
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "10.1:14550" "--out" "127.0.0.1:14551" "--console"
```
and
```
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5770" "--sitl" "127.0.0.1:5511" "--out" "127.0.0.1:14560" "--out" "127.0.0.1:14561" "--console"
```
I started 2 mavros nodes with fcu_urls of udp://127.0.0.1:14551@ and udp://127.0.0.1:14561@, respectively.

Furthermore, when looking at the topics `/mav0/mavros/mission/waypoints` and `/mav1/mavros/mission/waypoints` the correct waypoint tables are displayed.

So, I think these two planes are sharing waypoints in a way that doesn't seem entirely obvious to me.

I further tested this by removing all the mavros stuff and just running the `sim_vehicle.py` commands with a gazebo plugin. I loaded up two different waypoint tables `CMAC-Circuit.txt` and `CMAC-Grid.txt` using MavProxy. The results are in this [youtube video](https://youtu.be/CK51FROgs0c)

**Version**
Arduplane 3.8.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ * ] Plane
[  ] Rover
[  ] Submarine

**Logs**

For reference, I modified `gazebo-zephyr.parm` to support a 4 channel plane in gazebo. 

My modified `gazebo-zephyr.parm`
```
SERVO1_FUNCTION 4
SERVO2_FUNCTION 19
SERVO3_FUNCTION 70
SERVO4_FUNCTION 21
#SERVO1_FUNCTION = 77
#SERVO2_FUNCTION = 78
#SERVO1_REVERSED = 1
#SERVO2_REVERSED = 1

#RC3_MAX         2000
#RC3_MIN         1000
SERVO3_MIN      1000
SERVO3_MAX      2000
ARSPD_FBW_MIN   13
ARSPD_FBW_MAX   22
STALL_PREVENTION  1
LEVEL_ROLL_LIMIT  5
LIM_ROLL_CD     4500
LIM_PITCH_MAX   2000
LIM_PITCH_MIN   -2500
TRIM_ARSPD_CM   1600
PTCH2SRV_RMAX_UP  20
PTCH2SRV_RMAX_DN  20
AUTOTUNE_LEVEL   6
#RC3_TRIM        1000
#RC4_MAX         2000
TECS_APPR_SMAX  0
TECS_CLMB_MAX 5
TECS_HGT_OMEGA  3
TECS_INTEG_GAIN 0.1
TECS_LAND_ARSPD -1
TECS_LAND_DAMP  0.5
TECS_LAND_IGAIN 0
TECS_LAND_PDAMP 0
TECS_LAND_PMAX  10
TECS_LAND_SINK  0.25
TECS_LAND_SPDWGT  -1
TECS_LAND_SRC 0
TECS_LAND_TCONST  2
TECS_LAND_TDAMP 0
TECS_LAND_THR -1
TECS_PITCH_MAX  0
TECS_PITCH_MIN  0
TECS_PTCH_DAMP  0
TECS_RLL2THR  10
TECS_SINK_MAX 5
TECS_SINK_MIN 2
TECS_SPD_OMEGA  1
TECS_SPDWEIGHT  2
TECS_SYNAIRSPEED  1
TECS_THR_DAMP 0.5
TECS_TIME_CONST 5
TECS_TKOFF_IGAIN  0
TECS_VERT_ACC 7

NAV_CONTROLLER  1
NAVL1_XTRACK_I  0.02
NAVL1_DAMPING   0.75
NAVL1_LIM_BANK  0
NAVL1_PERIOD    19

RTL_RADIUS      50

PTCH2SRV_RLL    0.7

SR0_RAW_SENS    10
SR0_EXT_STAT    10

SR1_RAW_SENS    10
SR1_EXT_STAT    10

SR2_RAW_SENS    10
SR2_EXT_STAT    10

SR3_RAW_SENS    10
SR3_EXT_STAT    10

ARMING_CHECK    0
# we need small INS_ACC offsets so INS is recognised as being calibrated
INS_ACCOFFS_X   0.001
INS_ACCOFFS_Y   0.001
INS_ACCOFFS_Z   0.001
INS_ACCSCAL_X   1.001
INS_ACCSCAL_Y   1.001
INS_ACCSCAL_Z   1.001
INS_ACC2OFFS_X   0.001
INS_ACC2OFFS_Y   0.001
INS_ACC2OFFS_Z   0.001
INS_ACC2SCAL_X   1.001
INS_ACC2SCAL_Y   1.001
INS_ACC2SCAL_Z   1.001
INS_GYR_CAL      0
ARSPD_TYPE      0
ARSPD_USE       1
BATT_MONITOR    4
```

@dallinbriggs

khancyr, 2018-07-07:
Hello,
If you use sim_vehicle. py from the same directory, both planes will share the same eeprom.
Add --use-dir plane0, and --use-dir plane1 (or another name) in order to put them in their own directory !

DevonMorris, 2018-07-07:
Thanks so much! I'll close this issue then.

closed 2018-07-07 by DevonMorris