ArduPilot/ardupilot #8851: Rover: add autotune for steering and throttle control (at least)

rmackay9, 2018-07-09:
Although we have decent [tuning guides for steering](http://ardupilot.org/rover/docs/rover-tuning-steering-rate.html) and [speed](http://ardupilot.org/rover/docs/rover-tuning-throttle-and-speed.html) it could be faster/more accurate if we added an autotune mode that captures/learns things including:

- max rotation rate of the vehicle (ATC_STR_RAT_MAX)
- maximum lateral acceleration (TURN_MAX_G)
- steering rate control (ATC_STR_RAT_FF, ATC_STR_RAT_P, etc)
- throttle/speed control (CRUISE_SPEED, CRUISE_THROTTLE, ATC_THR_P, etc)

vkrm1612, 2022-04-05:
I would like to work on this issue for GSOC

utkarshraj779, 2023-02-27:
@rmackay9  Hlw I am Utkarsh Kumar from IIT Kharagpur 3rd year BTech student and I want to contribute to this gsoc 2023 project
please provide me information about where I should learn and grasp knowledge that should be benificial in the coming days. thankyou

fibonacci35813, 2023-03-17:
@rmackay9 I have been working on this for some time and would like to work on this issue for GSoC 2023. I will try to create a PR related to this in some time.

rmackay9, 2024-07-23:
This is done now with the introduction of QuikTune  https://ardupilot.org/rover/docs/quiktune.html

closed 2024-07-23 by rmackay9

bijuthaj, 2024-12-22:
Is there any plan to make it as a mode for rover? Could be useful for controller with no lua script support

rmackay9, 2024-12-23:
Hi @bijuthaj,

No plans for that at the moment I'm afraid.  It's certainly possible but some H7 based autopilots are relatively inexpensive and light so there are already hardware solutions out there i think.

bijuthaj, 2025-01-03:
Hi @rmackay9 
I was trying to add the auto tune as a mode for Rover. I have done the code changes by converting your lua script. I tested it with SITL and it seems working. But when I tested this with HITL on SpeedyBeeF405 board its getting crashed when switch to auto tune mode (17). 
The code is here in this repo.
https://github.com/bijuthaj/ardupilot/blob/rover_autotune/Rover/mode.h
https://github.com/bijuthaj/ardupilot/blob/rover_autotune/Rover/mode_autotune.cpp

[00000001.zip](https://github.com/user-attachments/files/18302194/00000001.zip)
[00000002.zip](https://github.com/user-attachments/files/18303296/00000002.zip)

 I couldn't download the crash dump its getting error. Not able to figure out the issue. Can you help me on this?

Below is the sequence of commands used from mavproxy.
MANUAL> rc 1 1500
MANUAL> rc 3 1500
MANUAL> arm throttle
MANUAL> mode circle
MANUAL> mode 17
CIRCLE> CIRCLE>
Mode(17)> Mode(17)>
Mode(17)>
Mode(17)>

rmackay9, 2025-01-04:
Hi @bijuthaj, ok, wow, that's quite impressive.  I don't think I'll be able to go through it all and debug the issue.  Instead I think you'll need to go through it and try and break down exactly where its failing.  Sorry about that, it's just that I've already got quite a few thing on my plate at the moment

bijuthaj, 2025-01-04:
Sure....I got it! Thanks @rmackay9 I will try to figure it out. I'm only starting with Ardupilot. So fining it bit difficult to debug issues of failure on  hardware without crashdump. Can you advice on the best way for debugging crash issues on the board? If anyone else here could help also welcome.

rmackay9, 2025-01-04:
@bijuthaj,

I think [testing with SITL](https://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) is a much better idea than trying to use HITL.

bijuthaj, 2025-01-04:
@rmackay9  I tested in SITL and it seems working fine. I could see tuning getting progressed and parameters getting saved. But on HITL I got few issues with same code. Main one was this method _get_steering_and_throttle_ is not available and getting compilation error. 
```
// get steering and throttle (-1 to +1) (for use by scripting)
bool Rover::get_steering_and_throttle(float& steering, float& throttle)
{
    steering = g2.motors.get_steering() / 4500.0;
    throttle = g2.motors.get_throttle() * 0.01;
    return true;
}
```
I moved the method to from Rover to mode_autotune.cpp class.  Now I could run the mode in HITL also but the tuning is not progressing. It's complaining and getting this message "RTun: increase steering ". It seems once I switched to auto tune mode from circle mode motor steering and throttle values are not getting updated. But the rover in the map is still doing circles. I think I'm missing something basic here.

bijuthaj, 2025-01-10:
Hi @rmackay9 
I have completed the changes and raised a PR https://github.com/ArduPilot/ardupilot/pull/29032 
When you get a chance can you or someone from team could review it?