ArduPilot/ardupilot #8855: FS_THR_ENABLED_CONTINUE_MISSION and not in AUTO mode causes LAND not RTL.

VDLJu, 2018-07-09:
### Bug report

In the radio failsafe documentation page it states that in case "Enabled Continue with Mission in AUTO" is used. 
> “Enabled Continue with Mission in AUTO” to allow the vehicle to continue with missions even if it takes the vehicle outside of RC range (not recommended). In all other cases the vehicle will RTL.

[Radio failsafe documentation](http://ardupilot.org/copter/docs/radio-failsafe.html)

Current solution causes a LAND if the drone is not in AUTO mode while the FailSafe is activated. This is not wanted behavior. Correct behavior would be RTL. The solution could be either set the RTL or add a parameter so it is possible to define the wanted behavior when not in a AUTO mode.

**Version**
3.6rc-2, but it should be happening in earlier versions too.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Any

**Logs**
Not relevant

VDLJu, 2018-07-09:
This is my first bug report and also first PR. 

This is my proposed fix: 
https://github.com/ArduPilot/ardupilot/pull/8854

rmackay9, 2018-07-09:
You are right!  This bug was introduced some time ago, well spotted!

VDLJu, 2018-07-09:
See https://github.com/ArduPilot/ardupilot/pull/8857. I did not know any other ways to make requested changes than recreate the PR.

OXINARF, 2018-07-11:
Fixed with #8857

closed 2018-07-11 by OXINARF