ArduPilot/ardupilot #8884: Add Speech/Voice output on pixhawk hardware

davidbuzz, 2018-07-11:
### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

Existing Audio Tones on the buzzer are not necessarily helpful in telling a user if/when there is a problem, and what that problem might be. Most user/s only recognise the "normal starup" tone, and any other tone is interpretted as "something went wrong, connect the gcs to get the details", which is not always helpful in field conditions, etc.

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._
I'd like error conditions to be able to tell you the actual error, rather than just a "special tone" that you need to memorise.  Eg "No GPS" or "Accel Calibration Needed" when spoken is much more meaningful than a bunch of beeps that you haven't heard before. 

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

This minimal speech library for the Arduino uses two high-speed PWM pins, and no DAC, so should ( in theory) be able to work on all the common stm32 platform/s, and is my suggested implementation, but if you find a better one, that's cool too. 
https://github.com/going-digital/Talkie

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here.

here's a demo of the audio quality (quite reasonable) : https://youtu.be/FCUUyzJx_f8?t=14s

rmackay9, 2018-07-11:
I think it's a good idea and I'm sure many users would love it. It's mostly just about finding someone to implement the changes in AP_Notify.

davidbuzz, 2018-07-11:
the code currently sets up two timers:
Timer 2 set up as a 62500Hz PWM
Timer 1 set up as a 8000Hz sample interrupt
and then has interrupt vector code that messes with their timing in order to get output, so it's currently the only bit of the implementation that would need to be reworked to run under chibios.    Might need @tridge's input here to see if chibios on stm32 has timer/s available to do this..?

zenuavsolutions, 2020-10-31:
I'd like to bring this old feature request back to life after the ap_notify melody contest ended last year. I think the autopilots with current processors would have enough resources and it would be a real benefit for the users.