ArduPilot/ardupilot #8890: Septentrio AsteRX-M2 GPS initialization regression

amilcarlucas, 2018-07-11:
### Bug report

Any Nuttx arducopter_v3.px4 version after commit 512939a839374801d00016b4704be442e16d9888 or fail to initialize the Septentrio GPS

**Issue details**
It gets stuck forever on the loop:
GPS 2: detected as SBF at 115200 baud
GPS 2: detected as SBF at 9600 baud
GPS 2: detected as SBF at 230400 baud
GPS 2: detected as SBF at 57600 baud
GPS 2: detected as SBF at 38400 baud
GPS 2: detected as SBF at 19200 baud
GPS 2: detected as SBF at 4800 baud
GPS 2: detected as SBF at 115200 baud

Using a Arducopter version previous to 2018-06-25 or reverting commit 512939a839374801d00016b4704be442e16d9888 solves the issue.

Some users at : https://discuss.ardupilot.org/t/master-branch-serial-issues/30755 also report a similar issue.

Tested with CibiOS and did not find any issues, this seams to a Nuttx/PX4 only issue

**Version**
master and tested a couple of versions before and after the troublesome commit

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk 2.1

**Logs**
_Please provide a link to any relevant logs that show the issue_

WickedShell, 2018-07-11:
`922a9762dfb0` px4-v3 AsteRx-M2 no issues. I'd check your setup, you seem to have a lot of init issues that I can't recreate.

amilcarlucas, 2018-07-11:
Are you using GPS1 and GPS2 as well?

WickedShell, 2018-07-11:
GPS on Serial3, Serial1 had a telem radio. The commit you are pointing at is specific to a single serial device, so interaction shouldn't matter.

jimovonz, 2018-07-11:
I get nothing intelligible out of telem1/2 (mavlink) using rover on current master (pixhawk/nuttx) . Exact setup/params using rover 3.4 is fine. Something is coming out of the port but I couldn't make sense of it at any baud rate lower at and than the 57000 baud the ports were set at. Also tried other baud rates

jimovonz, 2018-07-11:
I can confirm that it is the commit [cited](https://github.com/ArduPilot/ardupilot/commit/512939a839374801d00016b4704be442e16d9888) that is causing my issues. Clearing the read/write buffers in PX4UARTDriver::begin(); breaks the serial ports leading to no telemetry on telem1/2 and no GPS  (hal.consol is excluded). If I comment out the two lines clearing the buffers in current master, the problem goes away and everything works as expected

WickedShell, 2018-07-11:
@amilcarlucas @jimovonz can you post your full parameter set?

jimovonz, 2018-07-11:
@WickedShell Here you go
[jimovonz_rover.zip](https://github.com/ArduPilot/ardupilot/files/2186437/jimovonz_rover.zip)

tridge, 2018-07-11:
thanks! what sort of telem radios do you have? Do they have 4 wires (no flow control pins) or 6 wires (with flow control pins)
What radio settings do you have?

jimovonz, 2018-07-11:
@Tridge I use my own custom ESP32 based WiFi bridge for telemetry on Telem1 @ 500000bps. I disabled flow control for the telem ports (BRD_SER1_RTSCTS,0, BRD_SER2_RTSCTS,0) When testing to find the cause of this error I also tried a pair of cheap  433MHz radios on both telem1/2 with no luck. GPS is a S2525F8-RTK using NMEA @ 115200bps

tridge, 2018-07-11:
I haven't been able to reproduce the issue.
Can you see if this PR fixes it: https://github.com/ArduPilot/ardupilot/pull/8896
if it doesn't help, then do you have a logic analyser? I'd like to see a capture of the RX/TX lines on power up

tridge, 2018-07-11:
also note that there was an important bug fix for baudrate handling that went into master recently
can you make sure you have that fix:
https://github.com/ArduPilot/ardupilot/commit/5929b5fc33765a6bb37e96e432689d20eb5be3f4#diff-c530e0a830b4177e49320814d4a05245

jimovonz, 2018-07-11:
@tridge Yes that works for me - PR [#8896](https://github.com/ArduPilot/ardupilot/pull/8896) applied to current master fixes my problem. Cheers!

amilcarlucas, 2018-07-12:
This might have been fixed in master now by #8896 
@jimovonz please test it.

WickedShell, 2018-07-12:
@amilcarlucas he did see: https://github.com/ArduPilot/ardupilot/issues/8890#issuecomment-404341720

amilcarlucas, 2018-07-12:
Sorry, I did not :) I posted on a Firefox tab that I had opened yesterday :)

amilcarlucas, 2018-07-12:
I will test on my setup and close this issue once I can confirm that it also fixes the issue for me.

amilcarlucas, 2018-07-12:
I did test master, but it does not fix my issue :(

EViationGuy, 2018-08-28:
i can confirm same issue on arduplane 3.8.0 and 3.9.0, can try help resolving, anyone has an idea why it happens?

WickedShell, 2018-08-28:
@EViationGuy It's not the same issue @amilcarlucas is referring to. That issue would be present in 3.9, but it isn't present in 3.8 so if you have problems with both versions you must have a different root problem.

I can't reproduce the issues at all. At this point I'd like to see a logic trace from the serial line during boot/showing the problem because I'm afraid I just can't reproduce it at all.

EViationGuy, 2018-08-30:
Getting same issue with ChibiO so probably not a Nuttx issue.

`GPS 1: detected as SBF at 19200 baud
GPS 1: detected as SBF at 4800 baud
GPS 1: detected as SBF at 115200 baud
GPS 1: detected as SBF at 9600 baud
GPS 1: detected as SBF at 230400 baud
GPS 1: detected as SBF at 57600 baud
GPS 1: detected as SBF at 38400 baud
GPS 1: detected as SBF at 19200 baud
GPS 1: detected as SBF at 4800 baud
GPS 1: detected as SBF at 115200 baud
GPS 1: detected as SBF at 9600 baud
GPS 1: detected as SBF at 230400 baud
GPS 1: detected as SBF at 57600 baud
GPS 1: detected as SBF at 38400 baud
GPS 1: detected as SBF at 19200 baud
GPS 1: detected as SBF at 4800 baud
GPS 1: detected as SBF at 115200 baud
CubeBlack 00440040 3136510A 34313630
ChibiOS: 6686ee8a
ArduPlane V3.9.0 (9cc02ae1)`

WickedShell, 2018-09-02:
@EViationGuy That output is expected. At this point I think what I need to see is a logic analyzer trace from something like a [Saleae](https://www.saleae.com) or similar.

EViationGuy, 2018-09-04:
Hi

Got the setup ready with a logic analyzer on the relevant serial, do you want a recored from power up with both Cube and AsteRX-M2 on? or just the cube?

WickedShell, 2018-09-04:
@EViationGuy a record from power up on both the Rx and Tx lines to the GPS would be perfect!

EViationGuy, 2018-09-04:
Hi

Sorry, but i found out i had wrong wiring and bad configuration on the m2, once fixed the m2 configuration and wiring to pixhawk 2.1 worked with plane 3.9... 

sorry for the misleading information.

EViationGuy, 2018-09-06:
@amilcarlucas i can try other firmware as well if you want me to verify for you ... just let me know what you want to run.

same goes for all if you want me to check anything in order to close the issue.

amilcarlucas, 2018-09-07:
I have prepared a branch to test this:
https://github.com/ArduPilot/ardupilot/pull/9380

What FW version have you installed on your AsteRX M2

amilcarlucas, 2018-10-25:
@EViationGuy I'm using the latest available AsteRX M2 FW and I put #9380 on top of master and can reproduce the issue.

Can you also test that constellation?

amilcarlucas, 2018-10-25:
Yes, please build: https://github.com/amilcarlucas/ardupilot/tree/pr-disable-some-modules
@WickedShell closed #9380 so you will need to use my private pr-disable-some-modules branch.

Are you using a separated BEC for the septentrio power, or are you using the PH2.1 voltage?

EViationGuy, 2018-10-25:
it is powered from the PH2.1 GPS1 / 2 port...

EViationGuy, 2018-10-28:
> Yes, please build: https://github.com/amilcarlucas/ardupilot/tree/pr-disable-some-modules
> @WickedShell closed #9380 so you will need to use my private pr-disable-some-modules branch.
> 
> Are you using a separated BEC for the septentrio power, or are you using the PH2.1 voltage?

I am running Plane mostly, you want me to build for copter?

amilcarlucas, 2018-10-29:
yes, I'm getting problems with copter. It would be great if you could test that.

amilcarlucas, 2019-10-07:
I stopped having issues when I wrote the configuration into the M2 EEPROM. That way the GPS boots into the correct configuration and you bypass the (buggy) Ardupilot GPS auto-configuration step.

I do not fly that regularly, so I am not 100% sure that this fully solves the issue.

WickedShell, 2021-04-13:
I've done a fair amount of improvements to making the initialization support on SBF much more robust under a variety of circumstances, and can now reliably detect and configure the GPS, even after a hard factory reset of the unit, which is something that had never worked for me before. I'm going to close this now, as I think I've resolved the issue, if you test with master and still are encountering problems please let me know (or we can reopen this)

closed 2021-04-13 by WickedShell