ArduPilot/ardupilot #8914: Rover: improved error message when log download fails because vehicle is armed

rmackay9, 2018-07-14:
With Rover's it's quite common for users to try and download a log while the vehicle is armed.  The failure (viewed in MP) does not make it very clear that this is the issue.  I'm unsure if it's a mission planner issue or an ArduPilot issue but I'm adding it here in any case.

hdtechk, 2018-07-15:
There is a similar issue 1880 for mission planner

peterbarker, 2018-07-17:
Should we just issue a statustext?

tajisoft, 2018-10-27:
I think ArduPilot and Mission Planner are refusing to download logs in Armed state. What is the reason for denying downloads in Armed state? Is it dangerous?
https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/DataFlash_MAVLinkLogTransfer.cpp#L34
https://github.com/ArduPilot/MissionPlanner/blob/5b8c2a415d2d2fb5490b4be345842061df6ef179/ChangeLog.txt#L121

peterbarker, 2018-10-28:
On Sat, 27 Oct 2018, tajisoft wrote:

> I think ArduPilot and Mission Planner are refusing to download logs in Armed
> state. What is the reason for denying downloads in Armed state? Is it
> dangerous?

Only for anything underneath the vehicle :-)

Generally we avoid doing anything more complicated that writing to a file 
in dataflash while armed.  Doing so can cause signficiant scheduling 
issues - and if something goes wrong while working with the filesystem 
when downloading files the main loop may stop running - generally a bad 
thing.

PR #18512 (2021-09-08): AP_Logger: log transfer: add error message if armed
Fixes #8914

This is used on all vehicles, although the issue is just for rover.

Currently log related mavlink stuff will silently fail if armed. Will also fail if `!WritesEnabled()` but in that case AP will be printing "Initialising ArduPilot" every 5 seconds, so I don't think we need another message.
closed 2021-09-08 by tridge