ArduPilot/ardupilot #8915: Copter: follow with FOLL_OFS_TYPE = 1 (relative) does not initialise offsets properly

rmackay9, 2018-07-14:
The AP_Follow library supports two ways of specifying offsets:

- FOLL_OFS_TYPE = 0 means offsets (held in FOLL_OFS_X/Y/Z) are applied in NED frame from lead vehicle's position
- FOLL_OFS_TYPE = 1 means offsets are applied relative to the vehicle's heading

The issue comes when the FOLL_OFS_X/Y/Z values are left at their default of zero.  This is interpreted as meaning the offset to the lead vehicle should be initialised when the following vehicle enters Follow mode.  At the moment, the offset initialisation (done inside the AP_Follow::init_offsets_if_required() method always only works correctly if the offset-type is 0/NED.  This function should be enhanced to check the offset_type parameter and if it's relative then the offsets should be rotated appropriately based on the lead vehicle's heading.  The calculations to do this rotation can probably be borrowed from the bottom of the AP_Follow::get_offsets_ned() function.

hoangthien94, 2018-07-18:
In AP_Follow library the constant for relative vehicle's heading is currently the same as NED ([here](https://github.com/ArduPilot/ardupilot/blob/82aba6fe56ba143f5600b6a470e5805c1a562a41/libraries/AP_Follow/AP_Follow.cpp#L29)), am I correct? 

Although it does not affect how the code works since [get_offsets_ned](https://github.com/ArduPilot/ardupilot/blob/82aba6fe56ba143f5600b6a470e5805c1a562a41/libraries/AP_Follow/AP_Follow.cpp#L341) simple returns if  `_offset_type == AP_FOLLOW_OFFSET_TYPE_NED`, I think we should still change the constant to `1` for clarity. I'll give it a try.

rmackay9, 2018-07-19:
@hoangthien94, yes, the definition for AP_FOLLOW_OFFSET_TYPE_RELATIVE is certainly wrong.  If you PR that fix we will pull it right in.  Thanks for finding this!

hoangthien94, 2018-07-19:
No worries. I'm working on it as well as this issue, but right now I'm having another problem on master  https://github.com/ArduPilot/ardupilot/issues/8947.