ArduPilot/ardupilot #8920: Analog airspeed sensor on Matek F405-WING FC

sanyok84, 2018-07-14:
### Bug report

**Issue details**

I am really stuck on setting up analog airspeed sensor on Matek F405-WING with Ardupilot firmware from here http://firmware.ardupilot.org/Plane/latest/MatekF405-Wing/3. It works really well connected to RSSI pin on the board with ARSPD_PIN=15 and BATT_MONITOR=0, but as soon as i set BATT_MONITOR=4 and reboot the board - airspeed does not work anymore, thought battery monitor - does… This looks like some kind of a bug because battery monitor uses PIN 10 for voltage and 11 for current and i have no clue why airspeed on PIN 15 can’t be used after activating BATT_MONITOR. I activated analog RSSI through PIN 15 and it works with active battery monitor and displays data from airspeed sensor.

**Version**
ArduPlane V3.10.0-dev

**Platform**
[X] Plane

**Airframe type**
flying wing

**Hardware type**
F405-WING

**Logs**

sh83, 2018-07-15:
Really strange. Are you sure other ARSPD* params are ok?

sanyok84, 2018-07-15:
Yes all parameters are fine - it works really well with BATT_MONITOR=0 - i have callibrated the sensor on the ground with hair dryer (it is tested to produce 11m/s flow) and all is fine - airspeed hanges as soon as I start to blow on it, until i enable BATT_MONITOR=4. Reboot the board and Airspeed is always zero.

sh83, 2018-07-16:
Log mesages from mission planner would be helpfull. Like this https://goo.gl/images/iSaJ6q
Also log messages from telnet connected to port 2300 (mission planner prints all text messages to this port)

sanyok84, 2018-07-21:
Sorry for long reply. Here is the log from MP with BATT_MONITOR=4 (voltage/current sensor works, airspeed sensor constantly shows 0):

No IO Thread Heartbeat ()
u-blox 1 HW: 00080000 SW: ROM CORE 3.01 (107888)
EKF3 IMU0 buffers, IMU=11 , OBS=4 , dt=0.0200
MatekF405-Win 0045004E 3548500C 2037375
ChibiOS: c77e7465
ArduPlane V3.10.0-dev (d86dbee1)
Flight mode = 11
Failsafe. Long event on: type=2/reason=3
Airspeed[0] sensor calibrated
Flight mode = 5
Failsafe. Short event on: type=1/reason=3
Throttle failsafe on
GPS 1: detected as u-blox at 115200 baud
No IO Thread Heartbeat ()
-----------------------------------------------
Telnet localhost 2300
MPx!tR0Storage: failed at 0:1636 for 2 - re-init 1
-----------------------------------------------

I installed the latest build and still the same: 
BATT_MONITOR=4 - airspeed constantly show 0
BATT_MONITOR=0 - airspeed works fine

sh83, 2018-07-27:
It was suggested what maybe something wrong with params. Could you attach them?

IamPete1, 2018-09-02:
Were you able to resolve this, I am also having this issue on a F405-wing?

sh83, 2018-09-02:
I do not have this fc and analog sensor, so can not reproduce issue. I have looked into code, but it seems fine to me.

tridge, 2018-09-11:
I've reproduced the bug, and fixed it in this PR:
https://github.com/ArduPilot/ardupilot/pull/9403
thanks for the bug report!

closed 2018-09-11 by tridge