ArduPilot/ardupilot #8921: RE: Mixtures and Compiling w/ the CMDs sudo ./waf configure --board=blue and sudo ./waf

silver2row, 2018-07-15:
### Bug report

**While Compiling and While Configuring**

I think that w/out the iiod and the dependencies it needs, I found this out while using the cmd sudo ./waf configure --board=blue, the configuration is off. I typed that b/c the issues I found while using that cmd produced some odd results. 

These results were listed as lttng-ust not found and libiio not found. Then, I went on to produce this cmd: sudo ./waf. This cmd was to compile on my BBBlue. It is an Open Source board that has a Linux Distro on it.

Anyway...while compiling I noticed two issues:

There was a lack of:

    XML Validation Libraries

and this created a set of info. that stated that **XML libraries** will not be used.

    ../../libraries/AP_Terrain/TerrainIO.cpp: In member function ‘void AP_Terrain::open_file()’:
    ../../libraries/AP_Terrain/TerrainIO.cpp:176:36: warning: format ‘%u’ expects argument of type ‘unsigned 
    int’, but argument 5 has type ‘__gnu_cxx::__enable_if<true, double>::__type {aka double}’ [-Wformat=]
                  abs(block.lon_degrees));
                                        ^
    ../../libraries/AP_Terrain/TerrainIO.cpp:176:36: warning: format ‘%u’ expects argument of type ‘unsigned 
    int’, but argument 7 has type ‘__gnu_cxx::__enable_if<true, double>::__type {aka double}’ [-Wformat=]

    Here are my logs from journalctl -f.

    -- Logs begin at Sun 2018-07-15 03:17:22 UTC. --
    Jul 15 03:32:26 beaglebone systemd[1]: arducopter.service: Unit entered failed state.
    Jul 15 03:32:26 beaglebone systemd[1]: arducopter.service: Failed with result 'exit-code'.
    Jul 15 03:32:27 beaglebone systemd[1]: arducopter.service: Service hold-off time over, scheduling 
    restart.
    Jul 15 03:32:27 beaglebone systemd[1]: Stopped ArduCopter Service.
    Jul 15 03:32:27 beaglebone systemd[1]: Starting ArduCopter Service...
    Jul 15 03:32:28 beaglebone systemd[1]: Started ArduCopter Service.
    Jul 15 03:32:28 beaglebone arducopter[9645]: RCOutputAioPRU.cpp:SIGBUS error gernerated
    Jul 15 03:32:28 beaglebone systemd[1]: arducopter.service: Main process exited, code=exited, 
    status=1/FAILURE
    Jul 15 03:32:28 beaglebone systemd[1]: arducopter.service: Unit entered failed state.
    Jul 15 03:32:28 beaglebone systemd[1]: arducopter.service: Failed with result 'exit-code'.
    Jul 15 03:32:29 beaglebone systemd[1]: arducopter.service: Service hold-off time over, scheduling 
    restart.
    Jul 15 03:32:29 beaglebone systemd[1]: Stopped ArduCopter Service.
    Jul 15 03:32:29 beaglebone systemd[1]: Starting ArduCopter Service...
    Jul 15 03:32:29 beaglebone aphw[9684]: /bin/echo: write error: Operation not permitted
    Jul 15 03:32:29 beaglebone aphw[9684]: /usr/bin/ardupilot/aphw: line 6: 
    /sys/class/gpio/gpio80/direction: No such file or directory
    Jul 15 03:32:29 beaglebone aphw[9684]: /usr/bin/ardupilot/aphw: line 7: /sys/class/gpio/gpio80/value: 
    No such file or directory
    Jul 15 03:32:29 beaglebone systemd[1]: Started ArduCopter Service.

**Version**

    ArduCopter 3.5.5

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
    Hex

**Hardware type**
    Linux Board (BeagleBone Blue)

**Logs**
    I added the logs above to produce a finalized idea.

Seth

P.S. After compilation tonight, I am going to add the iiod and esp. libiio as the dependency under my Linux Distro. I have tried different kernels and different images so far, mostly Stretch 4.9 and 4.14.

silver2row, 2018-07-15:
Hello Again,

Also while compiling, this bout of messages appeared:

    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp: In member function ‘void 
    AP_GPS_SBP2::_sbp_process_message()’:
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:181:78: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
                 last_heartbeat = *((struct sbp_heartbeat_t*)parser_state.msg_buff);
                                                                              ^
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:181:78: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:186:76: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
                 last_gps_time = *((struct sbp_gps_time_t*)parser_state.msg_buff);
                                                                            ^
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:186:76: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:190:74: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
                 last_vel_ned = *((struct sbp_vel_ned_t*)parser_state.msg_buff);
                                                                          ^
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:190:74: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:194:74: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
                 last_pos_llh = *((struct sbp_pos_llh_t*)parser_state.msg_buff);
                                                                          ^
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:194:74: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:198:68: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]
                 last_dops = *((struct sbp_dops_t*)parser_state.msg_buff);
                                                                    ^
    ../../libraries/AP_GPS/AP_GPS_SBP2.cpp:198:68: warning: dereferencing type-punned pointer will break 
    strict-aliasing rules [-Wstrict-aliasing]

Seth

P.S. I am almost done compiling. I just thought since you and your people were in charge of this bug report, you needed to know what was happening to the software on a Debian Stretch Distro.

magicrub, 2018-07-15:
What version of gcc are you running?

silver2row, 2018-07-15:
Hello,

I am using:

    gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516

Seth

silver2row, 2018-07-16:
@magicrub ,

Hello Sir,

This is as far as I got recently. I listened to some people at imfatant on GitHub.com and I found some workarounds after goofing up.

    setup:307: bad header in setup - erasing
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver
    Sensor failure: Baro: unable to initialise driver

Seth

P.S. The Sensor Failure: error keeps printing on my terminal for now. I cannot stop it for some reason.

silver2row, 2018-07-20:
Hello Again Sir...

Please see here: https://pastebin.com/PZDLC68G. It is my terminal on PuTTY from the BBBlue. I cannot connect to the MP software yet for calibration.

I am close. i can feel it.

Seth