ArduPilot/ardupilot #8922: Request to support FrSky F.PORT protocol

flybabo, 2018-07-15:
### Feature request
FrSky introduced the F.PORT protocol which combines the SBUS and S.PORT telemetry into a single stream.  It also includes the RSSI info.
By using the F.PORT protocol, RX and FC can be connected through a single wire to a TX pin of UART when it operates as a half-duplex mode.

https://github.com/betaflight/betaflight/wiki/The-FrSky-FPort-Protocol
https://www.propwashed.com/frsky-fport/

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

Shackmeister, 2018-07-26:
+1
This would be a great addition!

wolfpackt99, 2018-08-01:
Agree. Full telem back to radio would be aewsome.

mclarenman01, 2018-08-11:
One less connection to make is always welcome!

davidngrc, 2018-08-23:
+1 one less wire, telemetry without extra hardware, nice.
anyone working on this?

x4FF3, 2018-09-01:
+1 already using it with inav builds. would appreciate it for my arduplane builds, too :)

yssaril, 2018-09-18:
+1 one less wires and and a free Uart if this can be added to the sbus/ppm-sum port

link to the FrSky anouncement [https://www.frsky-rc.com/frsky-betaflight-introduction-of-f-port-protocol/](https://www.frsky-rc.com/frsky-betaflight-introduction-of-f-port-protocol/)

utkinpol, 2018-09-30:
ditto, old request but if it needs more votes to get done - my vote is here as well. to have f.port would be convenient.

razorbac91, 2018-10-12:
I'm waiting it :)

maroquio, 2018-11-22:
I think this is one of the most important things to be supported to avoid telemetry issues. I hope you do it soon!

bugobe, 2018-12-04:
+1 for wanting this feature added. With the Frsky 900mhz radios becoming more popular, I think arduplane /f-port would be very popular for medium range aircraft.

ninja-zx11, 2018-12-11:
+1 for the need of this feature.Please add if possible!!

GregCovey, 2018-12-21:
F.Port would be a nice feature on some of these supported F4 boards with limited UARTs.

ignaciobfp, 2019-01-08:
This would be a nice addition, for simplicity and to bring telemetry to lots of users with less time/knowledge

sincoder, 2019-04-10:
+1 , with fport , telemetry become very simple !

PHMT, 2019-04-14:
Would be a helpful feature.  One less UART used (and available for something else) is very nice.

gregbiv, 2019-04-15:
+1 on this! I wish we can have it one day

sincoder, 2019-05-01:
+1 anyone could port FPORT from betaflight to Ardupilot ?

CpeDratDoPC, 2019-06-13:
I think this is one of the most important things to be supported to avoid telemetry issues.

Basti77, 2019-06-13:
Oh yes, that's my wish too. so I can save the smart port adapter cable and have one UART free. one less source of error.

maroquio, 2019-08-20:
+1 vote!

jace26, 2019-09-05:
+1 also, get rfd900x to support passthrough of f.port, and frsky to support f.port redundancy (does this exist yet?)

MFlghts, 2019-10-09:
+1 waiting hard for this

athertop, 2019-10-10:
Yes, this would be a great feature to have, saving an extra uart.

MoriXTV, 2019-10-10:
+1 F.Port would be amazing as we would be able to switch between iNav and ArduPlane just by flashing a new Firmware

Dronotique, 2019-10-14:
+1, should be a great feature

tridge, 2019-11-04:
sorry for the delay on this, I now have a F.Port RX and I plan to do an implementation soon

MFlghts, 2019-11-04:
Great news! Thank you very much!

mclarenman01, 2019-11-04:
> sorry for the delay on this, I now have a F.Port RX and I plan to do an implementation soon

Fantastic news! Thank you @tridge

variostudio, 2019-11-18:
It would be a nice feature

tridge, 2019-12-02:
PR to add FPort support here:
https://github.com/ArduPilot/ardupilot/pull/12945

closed 2020-01-01 by tridge

tridge, 2020-01-02:
FPort support is now in master.
Discussion and setup guide for testers here:
https://discuss.ardupilot.org/t/frsky-fport-support-testers-wanted/50669

PR #12945 (2020-01-01): Support FPort RC protocol
This adds support for the FPort RC protocol. It only does RC input so far, I will need to work with the FrSky SPort maintainers in ardupilot to plumb this into the SPort pass-thru support

Fixes #8922