ArduPilot/ardupilot #8925: Copter: support 4 lidar by default for Object Avoidance

rmackay9, 2018-07-16:
Copter-3.5 (and higher) supports using regular lidar for object avoidance but only 2 range finders are supported by default.  We have received a request to increase this to 4:

1. increasing the RANGEFINDER_MAX_INSTANCES definition to "4" for Copter only in AP_RangeFinder/[RangeFinder.h](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_RangeFinder/RangeFinder.h#L24).  Normally we do not want to have frame specific definitions in the .h file so it may need to be moved to RangeFinder.cpp.  That will have knock-on effects on the definitions of the state and drivers arrays.
2. testing that it works

This request came from [this Copter-3.6 discussion](https://discuss.ardupilot.org/t/copter-3-6-0-rc5-available-for-beta-testing/30812/32).

int-smart, 2018-11-20:
I am trying to solve this issue. Changing the code in header is direct, however considering we are trying to move this into the cpp file, I went through the cpp file. The CPP file has the var_info variable declared for RANGEFINDER_MAX_INSTANCES>3. Do I just need to calibrate the orientation default to front, rear, top and bottom directions and test it on different platforms or is their something that I am missing.

int-smart, 2018-11-27:
Hi. I added a PR#9913 for this Issue. Let me know if I missed anything.

rmackay9, 2020-01-13:
Copter-4.0 supports lots of lidar so closing this issue

closed 2020-01-13 by rmackay9