ArduPilot/ardupilot #8931: EKF bug, the gyrobias.z will drift and the gyroscale will enlarge

huangwen0907, 2018-07-17:
@priseborough  Hi, Could you please help me analyze the log, now I have a log tested and now I faced a text is  that: 
> I run the EKF2 with two IMUs and with the same one magnetometer.  now I found the the second EKF2 (`core[1]`) is healthy and the flying performance is correctly. but the first EKF2(`core[0]`)  will diverge when hold at one point with rotate the yaw direction quickly. it seems that the 'toilet' phenomenon will come out. 
**FYI**: If I use the EKF2 (`core[0]`) , the vehicle will also can not hold the altitude, it will up and down about a range of 1 meters. 
 I have done  the tests:
 - change the new imus, it will also core[0] is unhealthy and core[1] is healthy. 
 - exchange the imu[0] and imu[1] place, I found that it is still core[0] is unhealthy and core[1] is healthy. 
 - so if I use core[0], the flying behavior is not good as core[1].
all the core[0] and core[1] is run in the EKF2 of ardupilot.

This is the log: [ log link](https://github.com/huangwen0907/File/tree/master/log)  (sorry, I do not know how to post the .bin log here, so I create the file in my git repo)

Now I post some data as below.  
**EKF3.IMX vs EKF8.IMX**
[image]
**EKF3.IMY vs EKF8.IMY**
[image]
**NK1.GZ vs NK6.GZ**
[image]

huangwen0907, 2018-07-17:
now I am a little doubt this issues is related with https://github.com/ArduPilot/ardupilot/pull/7349 . @priseborough .  I am appreciate that if you can give me some info.

huangwen0907, 2018-07-18:
@liang-tang Tks, my code is rebased on the commit https://github.com/ArduPilot/ardupilot/commit/e2f5ae6a4cd3fd7bf7f9fde495bb8d835f442b4c

liang-tang, 2018-07-18:
you got a large NKF4.SM.

huangwen0907, 2018-07-18:
@liang-tang  yes, I noticed this is not normal, so I post the mag innovation data is also not normal. 
The `core[0]` and `core[1]` use the same one mag data（`OMPASS_USE, 1` and `COMPASS_USE1, 0`）, 
they use the different IMUs, so now I should doubt the gyro cause the problem. correct？

huangwen0907, 2018-07-18:
@liang-tang FYI： the NKF9.SM seems normal.

liang-tang, 2018-07-18:
two gyros raw data are almost same.

rmackay9, 2018-07-18:
I don't actually understand the EKF very well but I heard that one difference between the first and second EKF is that one consumes the compass's 2D vector while the other consumes a 3D vector.

huangwen0907, 2018-07-18:
@rmackay9 Thank you anyway.as it I see  the log 
[image]
The `red line` is updated only at the start time (consumes the 2D). then consumes the 3D all the time(`the IMX/IMY/IMZ`) except the last time.  So more debug maybe need .

priseborough, 2018-07-18:
Both EKF's start using heading fusion and then change to 3-axis fusion when clear of the ground so it is not due to a different fusion method. Both use the same magnetomer data. Magnetic heading innovations for both EKF's  are small large at the start - following plot is in deg

[image]

The initial yaw angles of the two EKF's are close and consistent with the innovations:

[image] and consistent wit he difference in innovations

There is evidence of some motor interference affecting both compasses but it is not severe

[image]

EKF2 Does have a susceptibility to buildup of roll/pitch errors due to sustained high rate yaw rotation but it normally affects both IMU's, and in this instance the error buildup is happening at a fairly low rate and in only half a turn. 

Where is the IMU and GPS positioned on your vehicle? I notice that the GPS position offset puts the GPs antenna 40cm to the right of the IMU 

Have you tried running EKF3?

When you swapped IMU's was that in replay (which is not accurate wrt data frame alignment) and did you try setting EK2_IMU_MASK to 1 and then 2 so that there is a single EKF instance running on different IMU data for each test?

huangwen0907, 2018-07-18:
@priseborough  Thank you very much for your reply.
>Where is the IMU and GPS positioned on your vehicle? I notice that the GPS position offset puts the GPs antenna 40cm to the right of the IMU

The IMU is in the central of the vehicle's body, but the GPS is placed on the one side of the vehicle's body. We have measured this value.  

>Have you tried running EKF3?

Sorry, have not yet, but now I will have a try right now.

>When you swapped IMU's was that in replay (which is not accurate wrt data frame alignment) and did you try setting EK2_IMU_MASK to 1 and then 2 so that there is a single EKF instance running on different IMU data for each test?

what I do the  replay is the step below:
>>./waf configure --board=linux
./waf build --target=tools/Replay
build/linux/tools/Replay -- MyLogFile.BIN

I am doubt this is not right,  I can not see the reply log created in the file 'logs/000001.BIN'. @priseborough How  can I do the right replay? I am very appreciate that you can give me some illustration.
I have done the  test about swapped the IMU is that: change the drivers indexs of IMUs, each time I found that the EKF2' `core[0]` is not right but the `core[1]` is right.

priseborough, 2018-07-18:
Which log are you trying to replay?

huangwen0907, 2018-07-18:
@priseborough I think the log [log_2_2018-7-10-18-16-14_replay.bin](https://github.com/huangwen0907/File/blob/master/log/log_2_2018-7-10-18-16-14_replay.bin) can do the replay as it record the data from the power on. 
This is the picture of the vehicle.
[image]

huangwen0907, 2018-07-18:
@priseborough  I have tested the EKF3, I found that it is the similar as run EKF2. this is the log I push https://github.com/huangwen0907/File/commit/4eae172757fcebeb34f33670c6a04db6bcbad012. 
**BTW:**
I post the plot data line.

[image]
[image]

huangwen0907, 2018-08-03:
@priseborough  I am very appreciated that  if you can give me some ideas or how to debug this bug: 
1. If I use the one compass and gps and almost same IMU datas (not the same sensors) for fusing, then the predict struct states of `core[0]` and the `core[1]` will be different.
2. If I found that the gyro bias is divergence as the picture below, how can I debug this problem and solve it. 
 
---
**FYI**
Such as the log and picture as below:
1. The vehicle crash at the start time of flying up.
2. my code is rebased on the release version :  https://github.com/ArduPilot/ardupilot/releases/tag/Copter-3.5.5
3. The log link: https://github.com/huangwen0907/File/tree/master/crash_log
[image]
[image]

priseborough, 2018-08-03:
What autopilot hardware are you using and what modifications have you made additional to 
https://github.com/ArduPilot/ardupilot/releases/tag/Copter-3.5.5 ?

priseborough, 2018-08-03:
The log 
https://github.com/huangwen0907/File/blob/master/log/log_2_2018-7-10-18-16-14_replay.bin does replay using latest master, but the file generated is not compatible with MAVExplorer and generates the following error message when loading 

"Skipped 22278 bad bytes in log at offset 5884583, type=(163, 149, 133)"

I will need to fix this before this issue can be progressed.

huangwen0907, 2018-08-03:
@priseborough 
>What autopilot hardware are you using and what modifications have you made additional to

I used the hardware of `px4-v2`. and we have add some function in the controller. And we do not modify anythings about the file of `AP_NavEKF2`. 

2> "Skipped 22278 bad bytes in log at offset 5884583, type=(163, 149, 133)"

I have the similar problem. I have to try to fix it but  I failed.

priseborough, 2018-08-03:
px4-v2 is a build target. What specific hardware are you using.

huangwen0907, 2018-08-05:
@priseborough  sorry to misunderstand what you means. 
My hardware is that:The MCU is `STM32F427` and with the imu sensor of ICM20602 as the first and the second imu is ICM20689. The compass sensor is IST8310.

huangwen0907, 2021-07-26:
closed, outdated

closed 2021-07-26 by huangwen0907