ArduPilot/ardupilot #8945: drone crash due to gps glitch

amitmanish, 2018-07-18:
Hello everyone,
My drone crashed while landing back in automode.  I analysed the drone logs and found gps glitch error.
I have ekf failsafe set to land. It should have landed but it crashed. Link to my logs is below. Can anyone look into the problem.

Link:    https://drive.google.com/open?id=18x1W4jkJUzniQH5NGy1Kq4uQHF66UDOw

Thanks 
###

EosBandi, 2018-07-19:
I does not looks like a crash caused by gps glitch. More like some onboard malfunction, ESC desync, or it can be a simple settle in power.
The copter destabilized earlier than any gps or ekf error happend. (As I see those errors caused by the crash itself).

rmackay9, 2018-07-19:
We shouldn't really be raising support items in the GitHub issues list.  This issues list is for confirmed bugs and enhancement requests.  Sorry to be strict but if we're not then our to-do list turns into another support forum (and we already have one of those).

closed 2018-07-19 by rmackay9