ArduPilot/ardupilot #8951: Mission Planner: no option for onboard mag cal with wifi connection

kd0aij, 2018-07-19:
### Bug report

only live cal available with wifi, only onboard when connected via USB

**Version**
master
[image]
[image]

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad X

**Hardware type**
Pixracer/ESP8266 with external I2C compass/GPS

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2018-07-19:
Looks like a valid issue but it should probably be raised in the [Mission Planner's issues list](https://github.com/ArduPilot/MissionPlanner/issues) so I'll close this one if that's OK.

closed 2018-07-19 by rmackay9

kd0aij, 2018-07-19:
result from that :) "how uptodate is your esp8266? this is likerly the esp8266 nothing to do with missionplanner"

auturgy, 2018-07-19:
Onboard Mag cal uses a mavlink2 message.
Older mavesp8266 is mavlink1 only.
That might explain the issue: updating the esp8266 firmware may well solve this.

Regards,

James

WickedShell, 2018-07-19:
> Onboard Mag cal uses a mavlink2 message.

No it doesn't, there are recent MAVLink 2 extensions to it, but the message is MAVLink 1.

kd0aij, 2018-07-20:
I guess this explains why I can start mag cal using QGC, but get no feedback (progress bar never moves).

WickedShell, 2018-07-20:
@kd0aij I'm sorry, maybe I missed something but I don't see an explanation why QGC would have a problem with mag cal.

rmackay9, 2018-07-20:
I'm guess but perhaps the old mavesp firmware didn't forward messages that it didn't recognise.  It's a guess but I suspect that using the new firmware described on the [ArduPilot ESP8266 wiki page](http://ardupilot.org/plane/docs/common-esp8266-telemetry.html) will fix the problem... and no amount of extra work in the GCSs will solve the problem.

kd0aij, 2018-07-20:
Is it possible that onboard cal feedback requires mavlink 2? That would explain the problem with QGC.

WickedShell, 2018-07-20:
@kd0aij nope, it is not possible, feedback is a MAVLink 1 message. https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L1363 The recent extensions are on the report message only (and are only needed for the orientation estimation)

kd0aij, 2018-07-20:
Then I guess this is evidence of incompatibility with the version of firmware on my old ESP8266. Should we just say it's not supported by ArduPilot?