ArduPilot/ardupilot #8956: copter 3.5.7 LIS3MDL on external compass no work

roque-canales, 2018-07-19:
hello HMC5383L is EOL now, and all ardupilot hardware manufacturing companies change the old HMC8353L by the LIS3MDL.

I plugged new gps from drotek that include this new magnetometer, and it seems to be not handeled by copter 3.5.7....

proficnc, 2018-07-19:
No, not all hardware manufacturers are Diy this, we are removing the LIS3MDL as well from the Here, and moving to the icm20948 which has the ak09916 inbuilt

auturgy, 2018-07-19:
Can you provide more details? Which firmware (nuttx or ChibiOS), a log, etc. The LIS3MDL is definitely supported (although it too is EOL).

rmackay9, 2018-07-19:
The LIS3MDL is supported in Copter-3.5.7 but I can imagine two possible reasons it's not found:

- the COMPASS_TYPEMASK parameter has been set to a non-zero value (try making it "0")
- the -v2 version of the firmware is being loaded by default by the ground station.  Try specifically downloading and then uploading "ArduCopter-v3.px4" from [this directory](http://firmware.ardupilot.org/Copter/stable/PX4/).

rmackay9, 2018-07-19:
I'm closing this because really it's a support issue, not a confirmed bugs or enhancement request.  This issues list is the to-do list for the developers and we need to be strict about this or it turns into another support forum and the devs will lose track of all they need to do.

closed 2018-07-19 by rmackay9