ArduPilot/ardupilot #8957: Optical Flow Simulation not wokring in SITL

trgiman, 2018-07-19:
### Bug report

**Issue details**
Dear all,

Simulated Optical Flow sensor is not receiving valid data. No matter if EKF2 or EKF3 is being used. (Parameter EK3_GPS_TYPE = 3 and RNG enabled), as a result UAV is not holding position in SITL. When I switch back to EK3_GPS_TYPE = 0 and use rc 3 1600 copter holds the position. 

When I do attempt to take-off using Dronekit GUIDED mode and not using simple takeoff method but custom takeoff method utilizing msg factory below

> msg = vehicle.message_factory.set_attitude_target_encode

 UAV can reach the height but can not hold the position in SITL in both cases when EKF3_GPS_TYPE 0 or EKF3_GPS_TYPE 3 UAV demonstrates same behavior(same for EKF2)

 Related discussion is at https://discuss.ardupilot.org/t/optical-flow-sudden-crash-log/30388/9 at the end of the thread.

In case this is not bug and I mis-configured something, please let me know. 

**Version**

3.5.5 ArduCopter

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

peterbarker, 2018-07-21:
On Thu, 19 Jul 2018, Martin Trgina wrote:

> Simulated Optical Flow sensor is not receiving valid data. No matter if EKF2 or EKF3 is being used. (Parameter EK3_GPS_TYPE = 3 and RNG enabled), as a
> result UAV is not holding position in SITL. When I switch back to EK3_GPS_TYPE = 0 and use rc 3 1600 copter holds the position.

I'm working on a branch here which tests this functionality in autotest. 
It certainly worked for me a week ago - the log contained messages 
indicating precision-landing was working.

https://github.com/ArduPilot/ardupilot/pull/8932/files#diff-205ba7eeea0bd19f1af46b8419add13eR1233

trgiman, 2018-07-21:
@peterbarker yesterday I did a real flight test and the behavior was same like in SITL. While copter is switched in to the GUIDED or GUIDED_NOGPS it will not hold position (gps physically removed - px4flow,lidar are on). During the real flight, when copter is in Loiter (GPS physically removed from copter).Copter holds position without any issues. All above applies for ArduCopter 3.5.5. 

I am going to do new tests with ArduCopter 3.6. According this link it should work in 3.6 - https://discuss.ardupilot.org/t/the-copter-does-not-takeoff-in-guided-mode-and-ekf-gps-type-3-optical-flow/16751/4   and Dronekit should even allow to use velocity commands and simple takeoff function.

peterbarker, 2020-11-25:
We have an autotest to ensure that flow generally works.

If there's still an issue around, please reopen this or a new issue.

closed 2020-11-25 by peterbarker