ArduPilot/ardupilot #8958: Improve Distance_sensors handling

khancyr, 2018-07-19:
### Feature request

Currently we trust distance_sensors message to be good... but it shouldn't. The message have min and max range, so the value should be checked within the range, and profit from other field too.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

gs-niteesh, 2018-08-26:
I would like to work on it but have no clue . I am new to open source contribution. I need ur help to solve this.

khancyr, 2018-08-27:
Hello,
The task is quite simple, if you look how we accept those message : 
https://github.com/ArduPilot/ardupilot/blob/cd5c59773bb5d4d4cdd4ba77437b3ede40ff8662/libraries/AP_Proximity/AP_Proximity_MAV.cpp#L58
https://github.com/ArduPilot/ardupilot/blob/cd5c59773bb5d4d4cdd4ba77437b3ede40ff8662/libraries/AP_RangeFinder/AP_RangeFinder_MAVLink.cpp#L49
We always consider that the value received are good. But we have no garantie ... We should check that the value is in min-max range and on accepted range from the backend (see on other rangefinder how it is done)
The id field should also be use to reject message from unkown sender.... (liked to https://github.com/ArduPilot/ardupilot/issues/6540)
I don't think we are using the covariance for now.
The timestamp could be use to discard old message .

gs-niteesh, 2018-08-29:
Why am i getting no response?

peterbarker, 2018-08-31:
@gs-niteesh Everybody's busy :-)

I suggest putting what you currently have up for comment - any extra sanity check there is probably going to be good, and adding more in the same spot should be easy enough after we can see what you've got so far.

gs-niteesh, 2018-08-31:
```
  void AP_RangeFinder_MAVLink::handle_msg(mavlink_message_t *msg)
    {
        mavlink_distance_sensor_t packet;
        mavlink_msg_distance_sensor_decode(msg, &packet);

        uint16_t p_current_distance = packet.current_distance;
        // only accept distances for downward facing sensors
       if (packet.orientation == MAV_SENSOR_ROTATION_PITCH_270 &&
            _packet.max_distance >= p_current_distance && packet.min_distance <= p_current_distance_)
       {
            last_update_ms = AP_HAL::millis();
           distance_cm = p_current_distance;
       }
       sensor_type = (MAV_DISTANCE_SENSOR)packet.type;
   }
```

Is this correct? And what else do i have to add

peterbarker, 2018-09-01:
@gs-niteesh This isn't a pull request :-)  Please ask on gitter if you need help creating one.

There are a few problems here.

We should not be setting the sensor type from the packet if we're not taking data from it (pre-existing problem)

The comment is now misleading because the code is checking more than just the rotation.

This should be rejigged to do an early-return when it isn't going to accept the data.

So invert the rotation check so if the rotation is NOT 270 then return without doing anything.

Add a second check which does an early-return if the measurement is outside the expected range.

Don't bother taking p_current _distance temporary - just take it from the structure twice.

Thanks!

gs-niteesh, 2018-09-02:
@peterbarker just created a pull request with all the changes that you mentioned. Pull request no: #9338

amanbora, 2018-10-11:
@peterbarker I would like to contribute to this issue.

peterbarker, 2018-10-13:
@amanbora @gs-niteesh has a PR in place for this issue.  I've just poked him on pushing that one forward.  There are plenty of other issues out there! :-)

khancyr, 2021-01-25:
done in https://github.com/ArduPilot/ardupilot/pull/15990

closed 2021-01-25 by khancyr