ArduPilot/ardupilot #8962: Limit Descend Speed in Manual mode (loiter, posHold)

mshajeehm, 2018-07-19:
Is there any possibility of addition of a new parameter named "Loiter_Max_Descend_Speed" in  **Copter**. This is to limit descend speed thus enabling smooth landing even if you pull the throttle stick to zero (accidentally or by choice) during mission. This will surely help in reduced number of accidents due to high landing speeds otherwise. 
The solution might be the incorporation of same concept as "Land_speed" used in RTL mode which limits the max descend speed.

`Plateform: Copter`

jmachuca77, 2018-07-19:
Loiter and posh old already have this. Please read http://ardupilot.org/copter/docs/altholdmode.html#altholdmode-controls

closed 2018-07-19 by jmachuca77

mshajeehm, 2018-07-19:
These adjustments are for climb and descend both. What if I want to climb at a faster rate with throttle to 100% however on throttle down to zero, descend slowly?

jmachuca77, 2018-07-19:
Pretty sure WPNAV speeds apply there

rmackay9, 2018-07-20:
It's included in Copter-3.6.  the params are renamed to PILOT_SPEED_UP and PILOT_SPEED_DN.