ArduPilot/ardupilot #8968: Error in generating romfs.o while building copter

eanswer, 2018-07-20:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

While running command `sudo ./waf copter` for board px4-v2, it reports error in generating romfs.o, detailed error is below: 
`[  0%] Generating romfs.o
Error copying directory from "/ardupilot/build/px4-v2/modules/PX4Firmware/src/firmware/nuttx/extras" to "/ardupilot/build/px4-v2/modules/PX4Firmware/tmp/custom-ROMFS/extras".
make[3]: *** [src/firmware/nuttx/romfs.o] Error 1
make[2]: *** [src/firmware/nuttx/CMakeFiles/romfs.dir/all] Error 2
make[1]: *** [src/firmware/nuttx/CMakeFiles/build_firmware_px4fmu-v2.dir/rule] Error 2
make: *** [build_firmware_px4fmu-v2] Error 2`

Actually, I didn't find extras in folder nuttx.

**Version**
Copter-3.6.7

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
Pixhawk

rmackay9, 2018-07-20:
I have seen this as well.  I've been getting around the problem by doing a "make clean" or "make px4-cleandep"

related discussion: https://discuss.ardupilot.org/t/error-while-compiling-arducopter-by-waf/31063

hoangthien94, 2018-07-20:
I think you can try delete the `tmp` folder (under `/ardupilot/build/px4-v2/modules/PX4Firmware/tmp`) and run **without `sudo`**:
```
./waf clean
./waf configure --board px4-v3
./waf copter
```

eanswer, 2018-07-20:
Still doesn't work for me, same error.

trgiman, 2018-07-30:
Hi.Any solution to this problem? I can compile without any issues on Ubuntu, but not on the MacOSX High Sierra.

lvale, 2018-07-30:
I would suggest to start using the Chibios builds 

` ./waf configure --board fmuv2
`
or 

` ./waf configure --board fmuv3`

```
Waf: Leaving directory `/Users/Luis/src/ardupilot/build/fmuv2'

BUILD SUMMARY
Build directory: /Users/Luis/src/ardupilot/build/fmuv2
Target          Text    Data  BSS     Total  
---------------------------------------------
bin/arducopter  905704  3900  193344  1102948

Build commands will be stored in build/fmuv2/compile_commands.json
'copter' finished successfully (1m29.998s)

```
@peterbarker I can replicate the reported issue, if you need help

eanswer, 2018-07-30:
@trgiman I haven't succeed yet, actually there is no extras generated during compiling, so I guess it failed in generating that.

trgiman, 2018-07-30:
@lvale. Thanks for comment. I am not very familiar with ChibiOS PX4Firmware build. Also not sure if it can work with —board aerofc-v1 (intel aero rtf). what @eanswer mentioned is right. But unfortunatelly I do not know yet where to change compile rules for PX4Nuttx — path to CMake file (apparently something is not configured well for osx since ubuntu works well)

lvale, 2018-07-30:
@trgiman Isn't the intel aero a linux type board ? We were discussing STM32 type boards which do run NuttX as an operating system. Chibios is an alternate, more efficient, operating system for the same STM32 boards, and ArduPilot will deprecate support for NuttX soon.

auturgy, 2018-07-30:
Aero RTF has an stm32f4 running the flightcode (usually), in addition to the Linux board.
There isn’t a ChibiOS port for it yet (or at least, nobody has pr’d one).

Regards,

James

auturgy, 2018-08-01:
> @ivale @auturgy.Yes I know, but my situation is for aero rtf, which is using stm32 based fc by default and there is no port for chibios yet.
>
> @lucasdemarchi are there any plans to port intel aero rtf for ChibiOS?

Thanks Martin - I understand your situation, was responding to Luis on the stm32 point.

It shouldn’t be difficult to do a ChibiOS port, it just hasn’t happened yet.

Regards,

James

lucasdemarchi, 2018-08-02:
@trgiman @auturgy I started a port to chibios, but I have no idea when I will finish it.

I only use Linux, I have no idea what's wrong with Mac/Windows.

trgiman, 2018-09-17:
Hi @lucasdemarchi have you been successful with porting Intel Aero RTF to Chibi OS? If yes I'll be happy to test it out on Mac OSX as well.

Eventually anybody from community was successful with building of (Nuttx based) 3.6 Copter for Intel Aero RTF on Mac OSX? Thank you.

trgiman, 2018-09-26:
@eanswer  the problem is related to version of cmake system starting with CMake 3.12 version (it is not MacOSX related issue). If you will switch to older version below 3.12 ( I am using 3.5.1 currently), compilation will succeed.See release notes of cmake 3.12 

https://cmake.org/cmake/help/v3.12/release/3.12.html#id1

say that cmake copy_diretory now fails when the source directory does not exist. The previous behavior was that the target directory was created empty. 

Apparently issue has been identified and solved by PX4Firmware. Please read at https://github.com/PX4/Firmware/issues/10368

I was not able to locate, where to fix it in Ardupilot PX4Firmware branch to allow usage of 3.12 cmake versions. Maybe here? https://github.com/ArduPilot/PX4Firmware/blob/8ca93fbd511a9ff544e11bae1184489b703c09fb/cmake/nuttx/px4_impl_nuttx.cmake#L332 I am not sure. Any suggestions @auturgy 

加油 加油

MikeCharikov, 2018-12-19:
@trgiman looks like you are right. CMake 3.12 doesn't work for px4-v3.  Cmake 3.10 does.

khancyr, 2019-04-05:
this isn't relevant on chibios, closing

closed 2019-04-05 by khancyr