ArduPilot/ardupilot #8974: Bat monitor (second battery) "battery_usedmah2" value is nonsense.

pompecukor, 2018-07-20:
### Bug report

We have 2 individual batteries in a setup. Both on their indivudial (mauch) sensors. Connected to port one and port two. Parameters are all identicall. Voltage and current valueas are correct, however battery_usedmah2 on the second battery is show is strange number. See screen capture below.
I have to say, I am not sure if its Arduplane mavlink or Mission planner that is the root.

Arduplane 3.10 latest beta
I do not think it is a version problem, we never tried it before and I dont think many people would either. As I said the other values are correct. This is a calculated value I believe. So hardware even is irrelevant.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

[image]

lvale, 2018-07-22:
@pompecukor both a bin log file and a tlog when this happens would be useful

pompecukor, 2018-07-22:
It is not in flight. Just bencs test. I think super easy to replicate. Just use whatever you have and set Bat 2 sensor 2 to be same as 1. (typically 3), then you get the sensors readings on both, they should be same. Then just set display on MP to show bat usedmah2.

WickedShell, 2018-07-22:
A tlog is needed here. I have a second battery monitor on an aircraft which is reporting total current to the GCS correctly (via the `BATTERY_STATUS` message anyways), and when I set up duplicate analog monitors in SITL it also behaves correctly. This might be a GCS problem, or a specific issue the analog input on some board type. The tlog is needed to asses that.

pompecukor, 2018-07-22:
ok, I will turn it on, make one and upload in a moment.
Yeah it could be MP. What GCS are you using?
Both current sensors are Mauch PL and AP is Cube.

pompecukor, 2018-07-22:
@meee1 maybe MP issue?

pompecukor, 2018-07-22:
I now uninstalled/reinstalles MP. Deleted config folder in docs. Just to make sure. Same problem still.

[image]

tlog incoming

pompecukor, 2018-07-22:
[2018-07-23 00-04-42.zip](https://github.com/ArduPilot/ardupilot/files/2217449/2018-07-23.00-04-42.zip)

So current, voltage, on bat1 and bat2 are reported fine. Only issue is with used current on bat2. See tlog attached.

pompecukor, 2018-07-22:
Here are all the bat parameters too.

[image]

WickedShell, 2018-07-23:
@pompecukor This looks to be a MP bug. I'm not really sure what MAVLink message it is using to grab the current consumed (the only one I know of is `BATTERY_STATUS` when the id is `1`). From the log file you posted:

```
2018-07-22 15:07:57.37: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 37, current_consumed : 20, energy_consumed : 16, battery_remaining : 99}
2018-07-22 15:07:57.37: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 36, current_consumed : 20, energy_consumed : 16, battery_remaining : 99}
2018-07-22 15:07:58.01: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 40, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:58.01: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 40, current_consumed : 20, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:58.50: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 42, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:58.50: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 41, current_consumed : 20, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:59.19: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 37, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:59.19: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 36, current_consumed : 20, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:59.73: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 35, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:07:59.73: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 35, current_consumed : 20, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:00.35: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 35, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:00.35: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 34, current_consumed : 20, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:00.93: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 39, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:00.93: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 38, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:01.45: BATTERY_STATUS {id : 0, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 43, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
2018-07-22 15:08:01.45: BATTERY_STATUS {id : 1, battery_function : 0, type : 0, temperature : 32767, voltages : [65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535], current_battery : 42, current_consumed : 21, energy_consumed : 17, battery_remaining : 99}
```

Given that the field appears to be correct here, and this indicates the battery monitor library is parsing it correctly, I believe this is a MP bug, and am closing it out here. (If it turns out MP is using some other field that we are encoding wrong definitely ping me back here)

closed 2018-07-23 by WickedShell