ArduPilot/ardupilot #9001: Copter: add Fence action to always LAND

rmackay9, 2018-07-24:
Copter's Fence feature lets the user specify the desired response by setting the FENCE_ACTION parameter.  The current options are "0" for "report only" and "1" for "RTL or LAND".  We should add a third option to always LAND.  This is apparently a legal requirement in Japan.

Here's some help on the implementation:

- extend the parameter description for the FENCE_ACTION parameter ([here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Fence/AC_Fence.cpp#L27))
- add a new definition to [AC_Fence.h](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Fence/AC_Fence.h#L19) called AC_FENCE_ACTION_LAND
- modify Copter's [fence.cpp](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/fence.cpp#L34) file to handle the new action so it switches to LAND mode.

khancyr, 2018-07-24:
I think that a stop and report would be good too. I case we use a companion computer, it is pretty usefull to just get a report that we breach the fence but stop at it !

rmackay9, 2018-07-26:
fixed in master thanks to @eskaflon

closed 2018-07-26 by rmackay9