ArduPilot/ardupilot #9006: Bootloader compile fail, uavcangen.py error

zdajun, 2018-07-25:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
git clone the  latest master ardupilot, configure and compile Bootloader, fail in uavcangen.py.: list index out of range

**Version**
copter 3.6

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Y6

**Hardware type**
Pixhawk

**Logs**
gary@DESKTOP-F0PJ3CI:/mnt/c/uav/ardupilot$ ./waf configure --board px4-v3 --bootloader
Setting top to                           : /mnt/c/uav/ardupilot
Setting out to                           : /mnt/c/uav/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : px4-v3
Checking for program 'arm-none-eabi-ar'  : /usr/bin/arm-none-eabi-ar
Using toolchain                          : arm-none-eabi
Checking for 'g++' (C++ compiler)        : /usr/bin/arm-none-eabi-g++
Checking for 'gcc' (C compiler)          : /usr/bin/arm-none-eabi-gcc
Checking for c flags '-MMD'              : yes
Checking for cxx flags '-MMD'            : yes
Checking for HAVE_CMATH_ISFINITE         : no
Checking for HAVE_CMATH_ISINF            : no
Checking for HAVE_CMATH_ISNAN            : no
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : no
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : no
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : no
Checking for header endian.h                   : not found
Checking for header byteswap.h                 : not found
Checking for program 'cmake'                   : /usr/bin/cmake
Checking cmake version                         : 3.5.2
Checking for program 'ninja, ninja-build'      : not found
Checking for program 'make'                    : /usr/bin/make
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 2.7.12
Checking for program 'python'                  : /usr/bin/python
Checking for python version >= 2.7.0           : 2.7.12
Source is git repository                       : yes
Update submodules                              : yes
Checking for program 'git'                     : /usr/bin/git
Gtest                                          : STM32 boards currently don't support compiling gtest
Checking for program 'arm-none-eabi-size'      : /usr/bin/arm-none-eabi-size
Benchmarks                                     : disabled
Unit tests                                     : disabled
Checking for program 'rsync'                   : /usr/bin/rsync
'configure' finished successfully (3.243s)
gary@DESKTOP-F0PJ3CI:/mnt/c/uav/ardupilot$ ./waf bootloader
Waf: Entering directory `/mnt/c/uav/ardupilot/build/px4-v3'
[1/4] Running Submodule update: PX4Firmware
[3/4] Running Submodule update: uavcan
[4/4] Running Submodule update: mavlink
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule path 'modules/uavcan/dsdl': checked out 'e96bd6f227cd3b3a487d6070eb044cba12338549'
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule path 'modules/uavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'

Cloning into 'dsdl'...
Cloning into 'libuavcan/dsdl_compiler/pyuavcan'...
Cloning into 'dsdl'...

Submodule 'pymavlink' (git://github.com/ArduPilot/pymavlink.git) registered for path 'pymavlink'
Submodule path 'modules/mavlink/pymavlink': checked out 'f55377066e5f5dba2c63b7a5d116454b1b0ce303'

Cloning into 'pymavlink'...

Submodule 'Tools/gencpp' (https://github.com/ros/gencpp.git) registered for path 'Tools/gencpp'
Submodule 'Tools/genmsg' (https://github.com/ros/genmsg.git) registered for path 'Tools/genmsg'
Submodule 'mavlink/include/mavlink/v1.0' (git://github.com/mavlink/c_library.git) registered for path 'mavlink/include/mavlink/v1.0'
Submodule 'src/lib/DriverFramework' (https://github.com/PX4/DriverFramework.git) registered for path 'src/lib/DriverFramework'
Submodule 'src/lib/ecl' (https://github.com/PX4/ecl.git) registered for path 'src/lib/ecl'
Submodule 'src/lib/matrix' (https://github.com/PX4/Matrix.git) registered for path 'src/lib/matrix'
Submodule 'src/modules/uavcan/libuavcan' (git://github.com/UAVCAN/libuavcan.git) registered for path 'src/modules/uavcan/libuavcan'
Submodule path 'modules/PX4Firmware/Tools/gencpp': checked out '26a86f04bcec0018af6652b3ddd3f680e6e3fa2a'
Submodule path 'modules/PX4Firmware/Tools/genmsg': checked out '72f0383f0e6a489214c51802ae12d6e271b1e454'
Submodule path 'modules/PX4Firmware/mavlink/include/mavlink/v1.0': checked out 'bf251caef3dbf610e5152115a2d81e00e54ab7e7'
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework': checked out 'd6ef668f4564e1c158910c52df7d71114bd2ca4e'
Submodule 'cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon.git) registered for path 'cmake/cmake_hexagon'
Submodule 'dspal' (https://github.com/PX4/dspal) registered for path 'dspal'
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/cmake/cmake_hexagon': checked out 'ba681fcd0d4c5ac95636dd8e9d8536dbd81beabd'
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/dspal': checked out '8f39699c6b79447f32f0714629da790bbf271aa5'
Submodule 'test/dspal_tester/cmake_hexagon' (https://github.com/ATLFlight/cmake_hexagon.git) registered for path 'test/dspal_tester/cmake_hexagon'
Submodule path 'modules/PX4Firmware/src/lib/DriverFramework/dspal/test/dspal_tester/cmake_hexagon': checked out 'ba681fcd0d4c5ac95636dd8e9d8536dbd81beabd'
Submodule path 'modules/PX4Firmware/src/lib/ecl': checked out 'f86ef3478219b4b1c31e17b8f308df4b69eda2f7'
Submodule 'matrix' (https://github.com/PX4/Matrix) registered for path 'matrix'
Submodule path 'modules/PX4Firmware/src/lib/ecl/matrix': checked out '95e3d7d6ce5274d9b7fec76c0093dc5eaa381abf'
Submodule path 'modules/PX4Firmware/src/lib/matrix': checked out '07fba8322a16cb2dac47e6a8ac7d21ec346313ff'
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan': checked out '702f6f05608885d3be993edb01394212bd2dade1'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule 'libuavcan/dsdl_compiler/pyuavcan' (https://github.com/UAVCAN/pyuavcan) registered for path 'libuavcan/dsdl_compiler/pyuavcan'
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/dsdl': checked out '9804a3e6972825586be252ce08dd899f44994b14'
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule 'dsdl' (https://github.com/UAVCAN/dsdl) registered for path 'dsdl'
Submodule path 'modules/PX4Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'

Cloning into 'Tools/gencpp'...
Cloning into 'Tools/genmsg'...
Cloning into 'mavlink/include/mavlink/v1.0'...
Cloning into 'src/lib/DriverFramework'...
Cloning into 'cmake/cmake_hexagon'...
Cloning into 'dspal'...
Cloning into 'test/dspal_tester/cmake_hexagon'...
Cloning into 'src/lib/ecl'...
Cloning into 'matrix'...
Cloning into 'src/lib/matrix'...
Cloning into 'src/modules/uavcan/libuavcan'...
Cloning into 'dsdl'...
Cloning into 'libuavcan/dsdl_compiler/pyuavcan'...
Cloning into 'dsdl'...

[ 5/14] Running uavcangen
[ 6/14] Creating build/px4-v3/ap_version.h
[ 7/14] CMake Configure px4
[ 8/14] PX4: Copying rc.APM to px4-extra-files/ROMFS/init.d/rc.APM
Waf: Leaving directory `/mnt/c/uav/ardupilot/build/px4-v3'
Build failed
Traceback (most recent call last):
  File "/mnt/c/uav/ardupilot/modules/waf/waflib/Task.py", line 286, in process
    ret = self.run()
  File "Tools/ardupilotwaf/uavcangen.py", line 24, in run
    input_dir = os.path.dirname(self.inputs[0].abspath())
IndexError: list index out of range

zdajun, 2018-07-25:
Same error when compile under linux

khancyr, 2019-04-05:
this is working now. Closing

closed 2019-04-05 by khancyr

NatPack84, 2019-04-11:
I'm having the same issue.  How did you solve it?

thanks