ArduPilot/ardupilot #9011: All vehicles: Change Motor Interlock naming to Throttle Hold

lvale, 2018-07-25:
The naming of Throttle Hold is commonly used on the RC universe, and is widely understood. ArduPilot should have the feature currently named "Motor Interlock" renamed to "Throttle Hold"

hdtechk, 2018-07-27:
Then we need to educate the RC universe.
In the industry I work in all the less expensive models that don't have cruise control have a throttle lock.    You can lock throttle at what ever percentage you want.  But when you go down hill the rpms and speed increase alot if you go up hill the rpms and speed decrease alot. Motor interlock seems more like it's meant to maintain the rpms its set at

hdtechk, 2018-07-27:
I was not say how it works. I was saying how inappropriate the term "throttle hold" is used in RC universe.

closed 2019-01-27 by lvale