ArduPilot/ardupilot #9022: crosscompiled binary failing to run.

awright424, 2018-07-26:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[ x ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
quad
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
pocketbeagle/pocket pilot
**Logs**
_Please provide a link to any relevant logs that show the issue_

``
debian@beaglebone:~$ sudo ./arducopter 
./arducopter: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.27' not found (required by ./arducopter) ``

This binary was crosscompiled on ubuntu 18.04. I am able to do this in a docker container running ubuntu 16.04 but I would like to know how to rectify this so I dont have to compile in docker.

WickedShell, 2019-06-27:
Closing as a stale support issue. Looks to be a missmatch between your libc versions though, for more support please go to discuss.ardupilot.org.

closed 2019-06-27 by WickedShell

monkeypants, 2019-08-29:
for future reference: the work around at https://github.com/mirkix/BBBMINI/issues/45 works for me.