ArduPilot/ardupilot #9037: BLHeli_32 pass thru is hit and miss on connection and flashing attempts

Kelly-Foster, 2018-07-28:
I have a Spedix GS35 BLHeli_32 ESC connected to a Matek F405-Wing, with latest BLHeli_32 firmware and very recent ArduPlane master on the flight controller.  Host computer is Windows 10.  I power up the aircraft with its flight battery, and start BLHeli_32 suite software, and then connect FC to laptop via USB cable.  Confirm COM port in Device Manager, and select that COM port for connection from the BLHeli_32 suite connection menus.  Connection through ArduPlane is hit and miss, and flashing new BLHeli_32 firmware is very hit and miss.  Last week the desire to flash newest BLHeli_32 firmware took eleven attempts before getting to 100%.  The % flashed failure point was random, ranging from < 10% to almost 100%.  Tridge suggested this was perhaps an IRQ issue, which I know nothing about.  Standing by to test any branches or methods that might solve this issue.

BLHeli_32 ESC programming (once connected) and ESC telemetry (especially to ChibiOS-supported OSD) is working very well, and a huge advance in drive train in-flight visibility and control.

Kelly

Attached:  [Screen shots of flash failures with BLHeli passthru 22jul18.pdf](https://github.com/ArduPilot/ardupilot/files/2238144/Flash.failures.with.BLHeli.passthru.22jul18.pdf)

lthall, 2018-07-30:
I have also had the same issue. I found the last ESC was the most likely to fail.

WickedShell, 2019-07-14:
I'm going to close this, as with the latest versions of ArduPilot master + BLHeli suite I've actually been very reliably making the connection. This wasn't previously true so I think something got better along the way. If people object to that conclusion then we can reopen this.

closed 2019-07-14 by WickedShell