ArduPilot/ardupilot #9043: Arducopter for Matek F722?

Seeelefant, 2018-07-29:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._
There is an excellent new FC from Matek http://www.mateksys.com/?portfolio=f722-std. Is there any chance to get Arducopter ported to that?

auturgy, 2018-07-29:
The matek F722 only has 512k of flash, so a normal, full featured port won’t fit.

Regards,

James

rmackay9, 2018-07-29:
Mateksys is apparently working on an STM F7 based flight controller with more RAM though..

closed 2018-07-29 by rmackay9

Seeelefant, 2018-07-30:
Ok, thank you. Just as my last question: There is the AnyFC F7: https://quadmeup.com/hands-on-anyfc-f7-from-banggood/ which has enough memory. Is that an interesting target?

auturgy, 2018-07-30:
I have a WIP hwdef for the anyfcf7. Just need to test it.

Regards,

James

auturgy, 2018-07-30:
The anyfc f7 doesn’t have an onboard barometer or compass though, so be aware of that - it needs external peripherals to function fully.

Regards,

James

Seeelefant, 2018-07-30:
Yes, unfortunately, I am aware of this.  However the board is highly interesting in terms of price and computational power and connectivity.....
Regards
--Wolfgang

Seeelefant, 2018-09-28:
Hi James,
can you send me you WIP hwdef?
Regards
--Wolfgang

mgrimes1907, 2018-09-29:
Did he ever get a version for Anyfc F7 going?

auturgy, 2018-09-29:
There’s a pr in for that

Regards,

James

auturgy, 2018-09-30:
With the f722, the mcu needs to be supported before the hwdef is useful.
UAVChallenge is over now, so I’ll look at it again.
Due to flash constraints (512k on the F722), it’s unlikely to be a very useful board, but might be ok for non gps stuff, or with a companion computer.

Regards,

James

Seeelefant, 2018-09-30:
Hi James,
it is an F745, it has 1MB flash at a rather modest price.
Regards
--Wolfgang

Seeelefant, 2018-09-30:
...the AnyFC F7 one :-)

auturgy, 2018-09-30:
Yup, sorry if the thread got confused!
A pr is in with a hwdef.dat for the AnyFC F7. It boots, connects etc, but I had to wait for an external barometer so I can properly test it.
I have that now, and UAV Challenge is over, so I hope to get it merged over the next week or so. Other testers welcome!
https://github.com/ArduPilot/ardupilot/pull/9246

The matek F722 I’m interested in doing, but as above it has some limitations and is a bit more work. 

Regards,

James

Seeelefant, 2018-09-30:
Yeps, originally the AnyFC F7 had a baro but it was optimized away by our chinese friends :-( , you can add an external baro via I2C or if you have very good soldering skills you will even find the baro pads on the board.

auturgy, 2018-09-30:
The ms5611, which the anyfc has pads for, is very susceptible to damage whilst soldering. I added an external bmp280 (actually I found a gps with a compass AND a baro, so am using that).

Regards,

James

rmackay9, 2019-03-13:
@andresiko, I've heard they have one available although I'm not sure if it is available from their website yet.  I'm afraid none of the AP developers have received theirs yet

Seeelefant, 2019-03-14:
There is another quite recent  F7 flightcontroller which looks interesting http://www.holybro.com/product/59, maybe it´s worth a try.

auturgy, 2019-03-14:
The Kakute F7 (both versions) are supported. They use an F745, like the AnyFC.

Regards,

James

andresiko, 2019-03-14:
Is there any F7 with dual camera selection available? not yet, isnt it? (I assume that the dual camera thingy is not yet suported by ardupilot, but also asume that will be a matter of time!)

auturgy, 2019-03-14:
I’m not aware of any F7 with dual camera inputs that isn’t an F722, so not on the ArduPilot radar just yet.

Regards,

James

amznHendes, 2019-04-03:
@auturgy Any progress on the F722 with Plane? Dying to try it out if it can fit on flash.

wiistriker, 2019-07-07:
Interested in Ardupilot on Matek F722 too

auturgy, 2019-07-07:
I haven’t looked at F722 for a while.
I’ve looked at what needed to happen to fit ArduPilot on the F412 for Skyrocket, and just don’t think people would be interested in ArduPilot without GPS functionality.
Someone else can have a crack of course, but I don’t see it getting to the top of my pile anytime soon

cheshira, 2019-08-13:
Wow! Matek F722-Wing is so popular now, but is not supported by Ardupilot? I saw this topic about a year ago and I saw that there are limitation with flash size, but that was not a stumbling-block matter.  So a year later I thought this matter is already solved, but no, it is not, unfortunately. Despite the grown popularity of Matek F722!

bavariaaa, 2019-09-05:
What is the problem? Can't we disable  parts of the firmware to make it fit for 512k of flash in the f722?