ArduPilot/ardupilot #9073: Unable to connect to simulated drone with Arducopter today (1st August)

VengerGB, 2018-08-01:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**

We're seeing issues when running locally with a simulated multirotor drone since https://github.com/ArduPilot/ardupilot/commit/b8030816b8f4ccd827ba6604fdc3ea8379d3ef59#diff-922dc91149ad093b7e6cf6bca88f8a76
...got checked In.

We're selecting 'Simulation' in Ardupilot and selecting 'Multirotor' from the firmware panel.

Files being downloaded and that the **arducopter.exe** window is spitting this out:

**invalid param RC7_OPTION in defaults file
AP_Param: Invalid param in defaults file**

And MavLink never connects. 'No heartbeat packets received' 

I might be conflating one or more issues here, apologies if that's the case.  We're seeing this problem consistently on various installs. 

**Version**
Using latest mission planner 1.3.57 1.3.6777.19762

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Simulated

**Hardware type**
Simulated

**Logs**
_Please provide a link to any relevant logs that show the issue_

VengerGB, 2018-08-01:
I'm assuming a huge amount here - basically this has died overnight.  If someone could help, we'd appreciate it.

peterbarker, 2018-08-01:
On Wed, 1 Aug 2018, Guy Pritchard wrote:

> invalid param RC7_OPTION in defaults file
> AP_Param: Invalid param in defaults file

That's my bad, sorry.

The problem comes about when using parameter files, when specifying a 
custom frame type to SITL.

These files will need updating to say "RCn_OPTION" in place of "CHn_OPT" 
or "CHn_OPTION" for n between about 7 and 12.

I've pushed in fixes for the frame parameter files in master.

rmackay9, 2018-08-02:
I think this is fixed now so closing.

closed 2018-08-02 by rmackay9

VengerGB, 2018-08-02:
That did the trick - thanks chaps.