ArduPilot/ardupilot #9084: Would like to ask how to export data from Drone Mission Planner After Drone has fly? I would like to know actual data what Drone has fly.

bhudhanin, 2018-08-02:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REQUEST WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please do a careful search before opening this, there are already a lot of feature requests**_
If there was a previous discussion in the forum, link to it

### Feature request

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]_

**Describe the solution you'd like**
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

auturgy, 2018-08-02:
User support questions need to be asked on the forum: discuss.ardupilot.org

closed 2018-08-02 by auturgy

rmackay9, 2018-08-02:
In any case, the copter wiki has a [Log section here](http://ardupilot.org/copter/docs/common-logs.html).