ArduPilot/ardupilot #9085: vehicle got a shock when change speed

liang-tang, 2018-08-02:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

**Issue details**
WPNAV_SPEED set to 500cm/s,  DO_CHANGE_SPEED set to 1m/s， when implement DO_CHANGE_SPEED，vehicle got a shock。
_Please describe the problem_

**Version**
_What version was the issue encountered with_
3.6.0 rc6
**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
SITL

**Logs**
_Please provide a link to any relevant logs that show the issue_
 

 

[2018-08-02 16-28-47.zip](https://github.com/ArduPilot/ardupilot/files/2252768/2018-08-02.16-28-47.zip)

liang-tang, 2018-08-02:
[image]

liang-tang, 2018-08-02:
[image]

liang-tang, 2018-08-06:
1.3.57 version test result
[image]
[image]

rmackay9, 2018-08-06:
@liang-tang, thanks for that evidence.  The vehicle will need to slow-down to the new speed of course, so it makes sense that it will lean back for a bit.  The question is whether that's too rough or not.

liang-tang, 2018-08-06:
sorry about the graph, it's pitch angel.

liang-tang, 2018-08-06:
All test in SITL, execute  LOITER_TURNS command.
**Copter 3.5.7**
[image]

**Copter 3.6.0 rc6**
  
[image]

**WP**
[image]

liang-tang, 2018-08-07:
@lthall  any idea about this？

RickReeser, 2018-10-26:
Is this the same issue as #8627?

IamPete1, 2021-09-12:
closing in favor of https://github.com/ArduPilot/ardupilot/issues/8627

closed 2021-09-12 by IamPete1