ArduPilot/ardupilot #9129: SERVO_RATE doesn't work when Q_ENABLE=1 on OMNIBUS F4 PRO V2

han8jun, 2018-08-06:
### Bug report

**Issue details**

Once the parameter Q_ENABLE set to 1, the servo both CH1 and CH2 continuously buzz, checked the PWM output rate was 400Hz. The parameter SERVO_RATE was default value 50 seems not work.em_

**Version**
ArduPlane V3.10.0-dev

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
tailsitter

**Hardware type**
OMNIBUS F4 Pro V2

**Logs**
none

sh83, 2018-08-08:
You probably have motor pwm6. Pwm 1,2,6 and pwm 2,3,4 share the same timers and have to have the same output rate. You could use digital servos, which ok for 400hz. Or i could build fw, that uses tim3 for pwm 1 and 2 (tim3 currently used by buzzer)

sh83, 2018-08-09:
Output rate is set to 400hz for motors to improve response. This can not be changed as i I now.
The simplest solution is to use pwm1  pwm2 for servos and pwm3 pwm4 for motors . With SERVOx_FUNCTION parameters you could reassign output function.

han8jun, 2018-08-09:
this solution works well, so nice!  Thank you @sh83

closed 2018-08-09 by han8jun

Marcos-Range, 2021-07-31:
Hi I have the same bug!
CAN you send to me a parameters and hardwares that you uses?