ArduPilot/ardupilot #9149: Rover/Boat: add seaweed removal failsafe

rmackay9, 2018-08-07:
This is a request to enhance the Rover crash check failsafe to add a new option to attempt to free the vehicle from seaweed if it gets entangled and can't move in Auto, Guided and RTL.

Rover currently has a crash check failsafe which can trigger when the vehicle flips over or (more relevantly for this issue) when in Auto mode if the vehicle is not making progress towards the waypoint.  The feature was designed for cars which can get stuck on things.

The maneuver involves spinning the motors forward and back intermittently for about a minute.

There are a couple of issues/questions to consider before implementing:

- does the maneuver work in practice?
- could it trigger accidentally and too often on boats that are perhaps fighting against a strong current?

This request came up on [this discussion](https://discuss.ardupilot.org/t/automatic-seaweed-removal-manoeuvre).

ncx94, 2018-08-20:
Another thing to consider is the current consumption caused by a stuck motor. We should implement a current limitation setting too to avoid burning the motor by sending it high current pulses for one minute

geofrancis, 2021-09-09:
Have you thought about looking at the RPM telemetry from speed controllers, most canbus controllers will report motor speed as well as blheli32 controllers when telemetry is enabled. then you wouldn't have to wait until the vehicle has dug its self into the Ground or overheated the motors with whatever is wrapped around the propeller since the flight controller would know immediately something has stopped the motor. Im specifically putting blheli telemetry on my boat specifically so I can monitor RPM and set current limits remotely for each motor in case one becomes stuck.